ucb-bar/chiseltest #2: Support peek/poking FIRRTL circuits/circuits with non-Chisel IOs

azidar, 2018-11-16:
It would be great if the tester could peek/poke FIRRTL circuits. Also, it would be nice to use it to test IOs which a FIRRTL transform added, which doesn't have a Chisel equivalent.

I'd recommend looking at FIRRTL's `CompleteTarget` class hierarchy, and hopefully use them to represent signals. We should also probably have normal Chisel references be converted to these Targets.

Thoughts?

edwardcwang, 2018-11-16:
This will also help a lot for re-using Chisel testers with post-synthesis/post-par netlists as well.

ducky64, 2018-11-16:
I think that makes sense, though I do want to keep the chisel-level circuit interface API. So some kind of shim for a FIRRTL circuit would be needed to provide the interface definition for testers2. It could be manually written, or auto-generated.

For post-syn/post-par, testers2 already uses a name mapping table to translate the circuit interface data structures (Chisel hardware objects) to the handle needed by the underlying tester. If the IO structure is mostly the same this should also extend to post-syn/post-par, with potentially more top-level flow options needed to select what (eg particular Verilog files / simulator) to test?

edwardcwang, 2018-11-17:
I think at the basic level just having a name mapping table from FIRRTL/Verilog to Chisel object will go a long way. The rest can be done in a modular/agile fashion.

ekiwi, 2021-08-17:
Peeking and poking firrtl circuits directly is now supported through the `Simulator` interface.

closed 2021-08-17 by ekiwi