ucb-bar/chiseltest #3: Support for Testing Combinational Ready Valid Controllers

stevenmburns, 2018-11-18:
The ready valid controllers I use have combinational paths through them. The decision on whether to perform a transaction depends on the valid and ready signals supplied to the controller in that cycle.
To test such a controller, our test benches check the combinationally generated output of the controller to also decided which transactions are to be performed. So there is a sequence like this:
1) Test bench pokes valid signals on DUT input channels and ready signals on DUT output channels.
2) The DUT generates ready signals on DUT input channels and valid signals on DUT output channels.
3) The testbench looks at these signals generated by the DUT and takes appropriate action.

What is the best way to write testbenches for these controllers?

Here is what I tried to do but fails because there are peeks in threads after there are pokes.
There are some more comments below the code.

````
package examples.testers2

import org.scalatest._

import chisel3.tester._

import chisel3._
import chisel3.util._

import scala.util.Random

class Join extends Module {
  val io = IO( new Bundle {
    val inp0 = Flipped( DecoupledIO( UInt(16.W)))
    val inp1 = Flipped( DecoupledIO( UInt(16.W)))
    val out = DecoupledIO( UInt(16.W))
  })

  io.inp0.nodeq
  io.inp1.nodeq
  io.out.noenq

  when ( io.inp0.valid && io.inp1.valid && io.out.ready) {
    io.out.bits := io.inp0.bits + io.inp1.bits
    io.inp0.ready := true.B
    io.inp1.ready := true.B
    io.out.valid := true.B
  }

}

class Tie extends Module {
  val io = IO( new Bundle {
    val inp = Flipped( DecoupledIO( UInt(16.W)))
    val out = DecoupledIO( UInt(16.W))
  })
 
  io.inp.nodeq
  io.out.noenq

  when ( io.inp.valid && io.out.ready) {
    io.out.bits := io.inp.bits
    io.inp.ready := true.B
    io.out.valid := true.B
  }

}

import chisel3.internal.firrtl.{LitArg, ULit, SLit}

object AlternativeTestAdapters {
  class RandomFlip( seed : Int, val num : Int, val den : Int) {
    val rnd = new Random( seed)  

    def nextBoolean() : Boolean = {
      val r = rnd.nextInt(den) < num
      println( s"Random flip: ${r}")
      r
    }
  }

  val rflip = new RandomFlip( 47, 12, 16)

  class BaseAdapter( clk: Clock, timeout : Int) {
    var nsteps = 0
    def step() : Unit = {
      if ( timeout > 0 && nsteps >= timeout) throw new Exception( s"Exceeded clock tick limit (${timeout}).")
      clk.step(1)
      nsteps += 1
    }
  }

  class ReadyValidSource[T <: Data](x: ReadyValidIO[T], clk: Clock, tossCoin : () => Boolean, timeout : Int = 0) extends BaseAdapter( clk, timeout) {

    x.valid.poke(false.B)

    def enqueue(data: T): Unit = timescope {
      while ( true) {
        if ( tossCoin()) {
          x.bits.poke(data)
          x.valid.poke(true.B)
          while ( x.ready.peek().litToBoolean == false) {
            step()
          }
          step()
          return
        }
        step()
      }
    }

    def enqueueSeq(data: Seq[T]): Unit = timescope {
      for (elt <- data) {
        enqueue(elt)
      }
    }
  }

  class ReadyValidSink[T <: Data](x: ReadyValidIO[T], clk: Clock, tossCoin : () => Boolean, timeout : Int = 0) extends BaseAdapter( clk, timeout) {

    x.ready.poke(false.B)

    def dequeueExpect(data : T): Unit = {
      while ( true) {
        if ( tossCoin()) {
          x.ready.poke(true.B)

          while ( x.valid.peek().litToBoolean == false) {
            step()
          }
          x.bits.expect(data)
          step()
          return
        }
        step()
      }
    }

    def dequeueExpectSeq(data: Seq[T]): Unit = timescope {
      for (elt <- data) {
        dequeueExpect(elt)
      }
    }

  }
}

import AlternativeTestAdapters._

class TieTestProbablyWrong extends FlatSpec with ChiselScalatestTester {
  val rnd = new Random()

  behavior of "Testers2 with Tie"

  it should "work with a tie" in {
    test( new Tie) { c =>
      val INP = IndexedSeq.fill( 100){ BigInt( 16, rnd)}
      val OUT = INP

      val source = new ReadyValidSource( c.io.inp, c.clock, rflip.nextBoolean)
      val sink = new ReadyValidSink( c.io.out, c.clock, rflip.nextBoolean, 2000)

      fork {
        source.enqueueSeq( INP map (_.U))
      }

      sink.dequeueExpectSeq( OUT map (_.U))
    }
  }
}

class JoinTestProbablyWrong extends FlatSpec with ChiselScalatestTester {
  val rnd = new Random()

  behavior of "Testers2 with Join"

  it should "work with a join" in {
    test( new Join) { c =>
      val INP0 = IndexedSeq.fill( 100){ BigInt( 16, rflip.rnd)}
      val INP1 = IndexedSeq.fill( 100){ BigInt( 16, rflip.rnd)}
      val OUT  = (INP0 zip INP1) map { case (x,y) => (x+y) & ((1<<16)-1)}

      val source0 = new ReadyValidSource( c.io.inp0, c.clock, rflip.nextBoolean)
      val source1 = new ReadyValidSource( c.io.inp1, c.clock, rflip.nextBoolean)

      val sink = new ReadyValidSink( c.io.out, c.clock, rflip.nextBoolean, 2000)

      fork {
        source0.enqueueSeq( INP0 map (_.U))
      }

      fork {
        source1.enqueueSeq( INP1 map (_.U))
      }

      sink.dequeueExpectSeq( OUT map (_.U))
    }
  }
}
````

I do like the way you can use implicit program counters in the threads so you don't need to write explicit state machines for random injection and extraction as well as waiting for the transaction to be performed.

So how do we make this work. Can we add a phase_step or some other action? Could it look like this:

````
object AlternativeTestAdapters {
...
  }

  class ReadyValidSource[T <: Data](x: ReadyValidIO[T], clk: Clock, tossCoin : () => Boolean, timeout : Int = 0) extends BaseAdapter( clk, timeout) {

    x.valid.poke(false.B)

    def enqueue(data: T): Unit = timescope {
      while ( true) {
        if ( tossCoin()) {
          x.bits.poke(data)
          x.valid.poke(true.B)
          phase_step()
          while ( x.ready.peek().litToBoolean == false) {
            step()
          }
          step()
          return
        }
        step()
      }
    }

    def enqueueSeq(data: Seq[T]): Unit = timescope {
      for (elt <- data) {
        enqueue(elt)
      }
    }
  }

  class ReadyValidSink[T <: Data](x: ReadyValidIO[T], clk: Clock, tossCoin : () => Boolean, timeout : Int = 0) extends BaseAdapter( clk, timeout) {

    x.ready.poke(false.B)

    def dequeueExpect(data : T): Unit = {
      while ( true) {
        if ( tossCoin()) {
          x.ready.poke(true.B)
          phase_step()
          while ( x.valid.peek().litToBoolean == false) {
            step()
          }
          x.bits.expect(data)
          step()
          return
        }
        step()
      }
    }

    def dequeueExpectSeq(data: Seq[T]): Unit = timescope {
      for (elt <- data) {
        dequeueExpect(elt)
      }
    }

  }
}
````
Every tick is made up of two phases. "phase_step()" will pass the first phase. "step()" will pass the second phase (or both phases if the first hasn't been based yet.)

chick, 2018-11-19:
@ducky64 I'm looking over this but I think this issue is going to need your attention

ducky64, 2018-11-19:
Yeah, I know. It's on my to-do list. As are a bunch of other things.

ducky64, 2018-11-25:
Yeah, your second approach is kind of what AdvancedTester currently does, but more understandable.

Overall, I think this is a tough problem. Here are my thoughts on possible solutions, though they all have drawbacks. Perhaps as a group we can come up with something more clever...

### Phases
Basically the longest standing solution to this problem and kind of a catch-all to thread order issues. Allows user-defined phases, where in each time step, all threads execute with phases as the primary partial order. Pokes / peeks between phases are not checked.

This provides a very coarse-grained solution to imposing a partial thread order, though it's unclear how scalable it is (or needs to be). Primarily, testers2 could define a few stock phases (eg main testdriver and monitor phases come to mind), though how additional user-defined phases would be ordered, particularly with regard to each other (which might not be aware of each other's existence), would be unclear. Potential solution: punt to the top-level test writer, who would specify constraints between phases that are ambiguous. But it's unclear how these solutions could be made re-usable, eg not specified for every test where they are used.

Note that your ReadyValidSink doesn't seem to neatly fit in the monitor phase, since it's also manipulating signals. I know UVM defines a bunch of stock phases (which I'm not familiar with, and arguably contributes to the learning curve).

Additionally, phases cannot go backwards in time, so when switching to an earlier phase, there must be some kind of time step advance (eg `clock.step()`) before any kind of circuit interaction. This would be checked dynamically, similar to inter-thread interaction checking. Still a gotcha, but at least noisy and safe.

### Thread Order Constraints
Kind of a more fine-grained version of phases, this instead allows users to specify ordering constraints between threads. By being very targeted, this avoids the scalability problem of phases, since the test writer isn't forced to lump a bunch of possibly different threads under a only-somewhat-close-sounding phase.

I haven't fully thought this through, so syntax and semantics are currently unclear. One possibility is for a thread-specific option:
```scala
val driverThread = fork { ReadyValidSource.enqueue(...) }
val receiverThread = fork.after(driverThread) { ReadyValidSink.dequeue(...) }
```

This suffers from the same problem as phases, where if you spawn a thread constrained to run before another thread executes, but after that thread has executed, it would incur a silent one-cycle delay. Potential solutions include:
- Only allowing threads to be constrained to run after an existing thread. Current thread ordering constraints means that any newly spawned thread should execute after all previous (and hence, reference-able in code) threads have run. It's unclear how much of a limitation not having a `before` constraint would be.
- Allow users to explicitly specify the one-cycle delay where necessary, similar in the phases proposition (where the system will check that the first thing done is a `clock.step()` call). Kind of annoying, in that it forces the test writer to spawn the thread one clock ahead, since a goal of testers2 is lightweight, simple, cycle-accurate testing.

Note that the current threading rules already impose a total thread ordering, so this constraint might just be a way of indicating what cross-thread accesses are safe (and won't cause an error).

### Example-specific issues
One issue with the example is that it doesn't appear either of the above solutions could be moved into infrastructure (because it interacts across a ReadyValid Sink and Source, and on different Bundles altogether) - so you'd have to attach a `phase(...)` or thread order constraint wrapper to each fork call, which is cumbersome though not onerous.

Alternatively, it might be possible to have a default phase associated with a wire / ReadyValidSink / etc, though I'm not sure about the generalizability of that abstraction.

ducky64, 2019-01-28:
It turns out I need this now. I've given this a bit more thought, and I think the best solution at a library infrastructure level is regions (borrowing hopefully familiar-to-some Verilog terminology for how events within a time slot are ordered; I previously called them ~phases~ but that's apparently confusing). The main change from above would be that region are defined as blocks, instead of being associated with a thread; this allows a thread to have driver and monitor actions.

There would be two regions per timeslot, main testdriver (default region) and monitor, in that order. Cross-thread operations are not checked between regions, since it forms a global synchronization barrier. Combinational logic continues to propagate instantly (observably, at least). There are no restrictions going forward through regions (main testdriver to monitor) - the thread will pause until the region executes, but going backwards requires a clock step (checked at runtime) before any further simulator interaction.

This very very loosely inspired by [SystemVerilog regions](http://www.sunburst-design.com/papers/CummingsSNUG2006Boston_SystemVerilog_Events.pdf), though [System]Verilog allows multiple iterations through regions in a timeslot where we don't. But that's probably to support simulating RTL and test stimulus generation in the same framework, whereas testers2 separates them and we don't need to worry about simulating RTL.

In this proposal, the modified [DecoupledDriver](src/main/scala/chisel3/tester/DecoupledDriver.scala) code would look like:
```scala
  def enqueueNow(data: T): Unit = timescope {
    x.bits.poke(data)
    x.valid.poke(true.B)
    region(Monitor) {
      x.ready.expect(true.B)
    }
    getSourceClock.step(1)
  }

  def enqueue(data: T): Unit = timescope {
    x.bits.poke(data)
    x.valid.poke(true.B)
    region(Monitor) {
      while (x.ready.peek().litToBoolean == false) {
        getSourceClock.step(1)
      }
    }
    getSourceClock.step(1)
  }

...

  def expectDequeue(data: T): Unit = timescope {
    x.ready.poke(true.B)
    region(Monitor) {
      waitForValid()
      expectPeek(data)
    }
    getSinkClock.step(1)    
  }

  def expectDequeueNow(data: T): Unit = timescope {
    x.ready.poke(true.B)
    region(Monitor) {
      expectPeek(data)
    }
    getSinkClock.step(1)
  }
```

One big question (as above) would be style conventions for where regions are needed, and avoiding the need for gotcha "you need to step the clock after you invoked this driver function". Above, functions always expect to be in the main testdriver region, and they return into the same testdriver region.

stevenmburns, 2019-01-28:
This looks good to me. Does it get rid of your issue expressed above: 'it would incur a silent one-cycle delay'?

ducky64, 2019-01-28:
Kind of: there's still a delay required, but the compromise is that if you don't put a clock step between switching to an earlier region and the next simulator interaction, it will error out.

Without a fundamental reworking of the abstractions towards a event driven approach like Verilog where a line can be run multiple times as sources are updated (which breaks the imperative programming abstraction), I don't think there's a perfect way around the going-backwards-in-time issue.

PR #15 (2019-02-07): Regions
Adds support for regions, which is a way to subdivide timeslots (atomic increments in simulator time) to allow combinational operations between threads to be resolved deterministically. This enables a passthrough decoupled interface to be driven using the DecoupledDriver abstraction.

This has `region(...)` as a blocking interface, which must be followed by a clock step if moving backwards. An alternative would be to have threads associated with an immutable region. To be discussed, as it would have a different set of tradeoffs, but this is being pushed for now.

Resolves #3
closed 2019-02-07 by ducky64