ucb-bar/chiseltest #5: [Request] Support Bundles in ReadyValid(Source|Sink)

nhynes, 2018-11-22:
Although it's possible to just define the queue ops on the DUT directly and poke manually, it'd be great if the `ReadyValid` wrappers could figure this out automatically!

ducky64, 2018-11-22:
You mean support ReadyValid where the bits are of Bundle type? This kind of works already, see the example at https://github.com/ducky64/ccc18-testing-intensive/blob/master/src/test/scala/T2GcdSpec.scala - specifically, the `enqueue` calls.

However, what is needed is a Bundle literal constructor, which I manually defined (kind of a hack) in https://github.com/ducky64/ccc18-testing-intensive/blob/master/src/main/scala/Gcd.scala

Writing a static annotation (like `@bundle` or something) that adds a literal constructor to a Bundle is a planned Chisel3 feature, though it hasn't seen significant work yet.

ducky64, 2018-11-22:
That being said, `peek` (and by extension `dequeue`) support is lacking for Bundles, since there's no programmatic Bundle literal constructor right now, but that's also planned.

nhynes, 2018-11-22:
I was actually about to fork the repo and add a `Map[String, Data]` overload to `peek`/`poke` which would allow setting the bundle elements individually. Is this a sustainable solution or should I just wait for `@bundle` or the like?

ducky64, 2018-11-22:
So what I've been doing right now is by manually writing Bundle literal constructors for the Bundles I need to work with. There are plans to create a Map[String, Data] Bundle literal constructor, which would form the basis of programmatic Bundle literal constructors needed for `peek` support.

I don't think the Testers2 ReadyValid drivers will be able to understand / detect custom peek / poke operations. They work on literal types of the wires being poked / peeked.

nhynes, 2018-11-22:
Thanks, that's a very helpful explanation! I'll just use what you've already made as a template and look forward to future versions of chisel+testers2 :)

closed 2018-11-22 by nhynes