ucb-bar/chiseltest #12: Support StrongEnum

edwardcwang, 2019-01-15:
$subject

zhuanhao-wu, 2020-01-07:
hi, I'm trying this framework in my project, and the project is heavily based on `StrongEnum` within Bundles. It would be quite helpful if `peek`/`poke` of `StrongEnum` is supported.

I wonder if you plan to add support for `StrongEnum` in recent snapshots?

Currently, I have a workaround in my local branch to allow me to test `StrongEnum` in `Bundles` that mimics the implementation of the `chisel-iotesters` with `Pokeable`.
It seems feasible by adding the `Element` version of `peek`/`poke`/`expect` etc in the backend.

ducky64, 2020-01-07:
Will be looking into this 'soon', along with other simple quality-of-life fixes like #105.

That being said if you have code that is PR-ready (architecturally clean and code quality clean) reviews probably happen faster...

ducky64, 2020-02-10:
I worked on this a while ago and ended up blocked on available Enum APIs (see #111)

@hngenc I think what I need is a programmatic way to get a literal given the enum hardware object (`EnumType`) and the number the enum corresponds to. Though `EnumFatory` has `apply(UInt)`, `EnumType.factory` is private (as it should be) - so maybe we can expose a specific API in `DataMirror`, like `DataMirror.intToEnum(int, chiselTypeOf(enumHardwareObject))`. Or, more broadly, `DataMirror.enumFactoryOf(enumHardwareObject)`?

ekiwi, 2021-08-17:
Let;s track this in #291

closed 2021-08-17 by ekiwi