ucb-bar/chiseltest #29: Expose TesterOptions / getTestOptions to programmatically set options

edwardcwang, 2019-04-16:
It can be in experimental or some other not-fully-supported API. Sometimes it is very useful to be able to set options (notably VCD dumping) programmatically as opposed to from the command line. Right now there isn't an easy way to programmatically set it since `TesterOptions` is private.

chick, 2019-04-16:
@edwardcwang in what way is `TesterOptions` private? It should be easy to read and/or write them as necessary

ducky64, 2019-04-16:
@edwardcwang can you give me an example use case? My thoughts are that things done in code are more permanent, while command-line operations are for temporary debugging flags like VCD dump.

If there is a reasonable use case I can open it up as an experimental API. Exposing options in a stable form is probably going to be very tricky.

edwardcwang, 2019-04-16:
TesterOptions is declared `private[tester] case class TesterOptions` so I can't use it directly from user code.

I want to be able to do something like
```
class MyTestSpec extends FlatSpec with ChiselScalatestTester with Matchers {
  override def getTestOptions: TesterOptions = {
    val test = scalaTestContext.value.get
    TesterOptions(test.name, writeVcd=<programmatically set VCD dumping>)
  }
  // rest of tests...
}
```

or even as an optional parameter to `test()` is fine as well, just some way I can set the settings

ducky64, 2019-04-16:
Yes, I understand that, but why doesn't command line options meet your use case? Do you want to dump VCDs for all runs (instead of only temporarily during debugging - and is this for all tests, or just a few)? Do you not use sbt and need command line options exposed some other way? Do you have a layer on top of testers2 that propagates the vcd dump switch from higher?

You're asking for a mechanism - and a really general / powerful one that is likely to be unstable. I want to be a bit more confident that this is the right mechanism, not just the one that appears most convenient right now.

edwardcwang, 2019-04-16:
I need it for `not use sbt and need command line options exposed some other way` and `layer on top of testers2 that propagates the vcd dump switch from higher [up in a flow]`.

I'm willing to accept breakage and it's totally cool if it's not the final API. I'd rather help test testers2 than go back to testers1 especially for a new project...

PR #31 (2019-04-19): Add experimental with[Option] to run, high/low() on clock
Resolves #29 

Perhaps unsurprisingly, this also went deep down the refactor rabbit hole.

- Refactors `run(dutGen => T)` to return a TestBuilder, which can be chain applied to run a test.
- Adds an experimental implicit conversion on the ChiselScalatestTester TestBuilder to allow `.withExecOptions`, `.withTesterOptions`. See the AsyncResetRegTest for an usage example.
- Removes the `run(dutGen, execOptions)` methods, so you have to import experimental to specify options.
  - Note: TestBuilder is not provided for RawTester (so there's n [... cut]
closed 2019-04-19 by ducky64