ucb-bar/chiseltest #35: Peeking zero-width I/Os

edwardcwang, 2019-04-30:
While I know that zero-width I/Os don't really exist, it might be nice as syntactic sugar to be able to peek (or even dummy poke?) zero-width wires and have it return 0. This is especially useful in generators where parameters might be set in a way such that some wires end up with zero-width.

This might also be a [treadle](https://github.com/freechipsproject/treadle) issue... @chick ?

Not super high priority since it can be worked around with some if statements.

Currently it fails with a slightly cryptic message:

```
java.lang.AssertionError: assertion failed: Error: getValue("zero_width_io_name") : argument is not an element of this circuit
  at scala.Predef$.assert(Predef.scala:219)
  at treadle.executable.ExecutionEngine.getValue(ExecutionEngine.scala:165)
  at treadle.TreadleTester.peek(TreadleTester.scala:245)
  at chisel3.tester.TreadleBackend.peekBits(TreadleBackend.scala:90)
  at chisel3.tester.TreadleBackend.expectBits(TreadleBackend.scala:99)
  at chisel3.tester.package$testableData.expectWithStale(package.scala:72)
  at chisel3.tester.package$testableData.expect(package.scala:91)
```

chick, 2019-04-30:
Hmm. This would be easy to do, perhaps as a mode where you can poke or peek anything you want without erroring. But treadle only has symbols for things that have survived all the way to low firrtl. so it would not otherwise know that the signal is zero length, it just simply isn't there, right?

edwardcwang, 2019-04-30:
Yeah, that's the tricky part methinks... I guess this is the inverse of #2.

ducky64, 2019-04-30:
Perhaps zero-width signals shouldn't disappear in low-FIRRTL? Even in Chisel frontend we kind of have a distinction between zero-width signals and optional Bundle fields, and zero width signals could be considered to be a form of design intent?

edwardcwang, 2019-04-30:
Well, zero-width wires are also a Chisel/Chirrtl thing and don't actually exist in hardware...

chick, 2019-04-30:
Annotations

edwardcwang, 2019-04-30:
[image]

ekiwi, 2023-01-11:
This should be fixed by: https://github.com/ucb-bar/chiseltest/commit/7d98ab58459832607361d045a90bdbccfe265297

closed 2023-01-11 by ekiwi