ucb-bar/chiseltest #39: waitFor construct

ducky64, 2019-05-19:
This would have equivalent arguments to what is supported by expect. Could provide better debugging information on timeouts, and could be more efficient in simulation since threads can be put to sleep.

In the future this API could be expanded with BitPat matches and stuff.

chick, 2019-05-19:
@ducky64 
Is there a use case for waiting for first of a list of events, something like
```
waitFor(cond1) { thunk1 }
.orElse(cond2) { thunk2 }
...
```
Or would you suggesting handling this by having separate forks each waiting for one thing, which might handle it but I am not sure how you would end the other forks once the first to fire came in.

ducky64, 2019-05-19:
There might be, but I'd like to see some concrete use cases first. That almost would seem like a different construct entirely. I'd also expect the waitFor to be on something like a valid signal, then the rest of the code could handle paths by peeking on other signals like bits.

A second question about waitFor is how to specify the clock edge it aligns on, or whether that should be specified at all (as in, waitFor stops blocking immediately after a match, though checking for thread ordering / inter-thread dependency issues). waitFor also shares some conceptual overlap with `clock.step(...)`, in that it's basically an edge-sensitive waitFor, eg `clock.waitFor(rising)`.