ucb-bar/chiseltest #49: how to install it

pp-hub, 2019-07-29:
I git it but when run sbt "testOnly chisel3.tests.BasicTest -- -DwriteVcd=1" there are a lout of errors like 
`object stage is not a member of package chisel3`
I think maybe i should install the latest chisel3 ,but in sbt is already  "chisel3" -> "3.2-SNAPSHOT", so what's  the problem

ducky64, 2019-07-29:
Are you using testers2 from git master, or from published dependencies?

If from git master, it currently relies on some very recent chisel3 features, and is incompatible with the current chisel3 snapshot published. We're hoping to publish an updated snapshot of chisel3 and friends soon,

In the meantime, if you're adventurous, you can build chisel3 and friends using publishLocal, and testers2 should build against locally published versions of chisel3/firrtl compiled from master.

If the latter, the testers2 snapshot might be incompatible and needs to be updated. If that's the problem I can update it.

pp-hub, 2019-07-29:
oh, i use the git master... so maybe i should try pubilshlocal . thank you very much and would you please tell me how to do that. if there is something like tutorial 
i git the chisel3 tester2relaxed  branch ,and then do what

kammoh, 2019-07-29:
Use master branch for everything. Checkout and publishLocal in order: firrtl, treadle, firrtl-interpreter (if needed), chisel3, chisel-testers2.
I’d also recommend cleaning up ivy2 cache/local beforehand.

ducky64, 2019-07-29:
Or, if you want to use a previous version of testers2 that is compatible with the currently published chisel/firrtl snapshots, you can use commit c1793bb21e152f2d3a420d28f3beaf054ed8b3a9

The main changes since then are the addition of the Verilator backend and internal refactors on how it interacts with chisel and firrtl.

pp-hub, 2019-07-30:
thanks a lot , i'll have a try

ducky64, 2019-08-04:
By the way, chisel/firrtl snapshots were updated recently and testers2 master should compile against those now.

closed 2019-08-04 by ducky64