ucb-bar/chiseltest #71: Access to a clock counter

schoeberl, 2019-09-20:
For simulation, it would be convenient to have access to a clock cycle counter. Adding a counter into the design and then routing it to an output is not the best workaround.

ducky64, 2019-09-20:
So under the current single-clock semantics, such a counter is fairly straightforward, but under multiclock semantics with some notion of simulator time, it might become less clear. 

For those, I was thinking:
- Probably have a way to inspect simulator time, defined in seconds
- Maybe have a way to count rising edges of clocks since the beginning of simulation, though this will need a treadle API (@chick ?) since not all clocks are necessarily under the user's control and testers2 may not be able to track everything. Note that we currently only track clocks that we know about (threads are blocked on), so rising-edges-since-beginning-of-simulation, while intuitive, might not be practical to implement. An alternative might be to have a ClockCounter / etc construct that you declare, and it tracks the number of rising edges since instantiation, but it'd be another special construct...

edwardcwang, 2019-09-20:
Seems like the best way would be a per-clock API/object instantiated by the user...

schoeberl, 2019-09-23:
Yes, just per clock should do it. Something like: dut.clock().ticks()

schoeberl, 2020-08-07:
Continue the discussion here: about ticks under reset: start counting after reset is released. Just like you would have a hardware counter:
```
cnt = RegInit(0.U(32.W))
cnt := cnt + 1.U
```

That's what I do as workaround.

PR #510 (2022-03-23): Add a getStep method for clock step "peeking"
Added a (naive) solution to track the clock steps that is not dependent of the backend.

The solution could track multiple clock domains thru a mutable map of values but the test backend apparently doesn't support it (a test has been written but ignored).

Fixes #71 , #398
closed 2022-03-23 by ekiwi