ucb-bar/chiseltest #72: 0.1-SNAPSHOT not in sync with chisel3 3.2-SNAPSHOT

ssteffl, 2019-09-30:
I think the SNAPSHOT of chisel-testers2 is not in sync with the other SNAPSHOTS

I cloned the master branch of https://github.com/freechipsproject/chisel-template/

I am running into the error below, and inspecting release dates of the SNAPSHOTS, it looks like chisel-testers2 was released 2 days before the latest chisel3 on https://oss.sonatype.org/content/repositories/snapshots/edu/berkeley/cs/.

It looks like MultiIOModule moved from experimental in this PR: https://github.com/freechipsproject/chisel3/pull/1162

Is the best course of action to just download all the repos and build/publish them locally?

`sbt:mytest> test:runMain gcd.GCDRepl
[info] Updating ...
[warn] Choosing sonatype-snapshots for edu.berkeley.cs#chisel3_2.12;3.2-SNAPSHOT
[warn] Choosing sonatype-snapshots for edu.berkeley.cs#firrtl_2.12;1.2-SNAPSHOT
[warn] Choosing sonatype-snapshots for edu.berkeley.cs#chisel-iotesters_2.12;1.3-SNAPSHOT
[warn] Choosing sonatype-snapshots for edu.berkeley.cs#firrtl-interpreter_2.12;1.2-SNAPSHOT
[warn] Choosing sonatype-snapshots for edu.berkeley.cs#treadle_2.12;1.1-SNAPSHOT
[warn] Choosing sonatype-snapshots for edu.berkeley.cs#chisel-testers2_2.12;0.1-SNAPSHOT
[info] Done updating.
[info] Compiling 2 Scala sources to /k/tmp/chisel-test3/target/scala-2.12/classes ...
[warn] there were 5 feature warnings; re-run with -feature for details
[warn] one warning found
[info] Done compiling.
[info] Packaging /k/tmp/chisel-test3/target/scala-2.12/mytest_2.12-0.0.1.jar ...
[info] Done packaging.
[info] Compiling 3 Scala sources to /k/tmp/chisel-test3/target/scala-2.12/test-classes ...
[error] /k/tmp/chisel-test3/src/test/scala/gcd/GcdTesters2.scala:25:5: Symbol 'type chisel3.experimental.MultiIOModule' is missing from the classpath.
[error] This symbol is required by 'type chisel3.tester.ChiselScalatestTester.T'.
[error] Make sure that type MultiIOModule is in your classpath and check for conflicting dependencies with `-Ylog-classpath`.
[error] A full rebuild may help if 'ChiselScalatestTester.class' was compiled against an incompatible version of chisel3.experimental.
[error]     test(new DecoupledGcd(16)) { dut =>
[error]     ^
[error] /k/tmp/chisel-test3/src/test/scala/gcd/GcdTesters2.scala:25:10: type mismatch;
[error]  found   : gcd.DecoupledGcd
[error]  required: T
[error]     test(new DecoupledGcd(16)) { dut =>
[error]          ^
[error] two errors found
[error] (Test / compileIncremental) Compilation failed
[error] Total time: 18 s, completed Sep 30, 2019, 4:40:07 AM
`

Martoni, 2019-11-06:
I have exactly the same problem : 
```
sbt "test:testOnly wbplumbing.WbInterconOneMasterSpec"
[info] Loading project definition from /usr/local/projects/videautom/gateware/hw_validation/WbPlumbing/project
[info] Loading settings for project wbplumbing from build.sbt ...
[info] Set current project to wbplumbing (in build file:/usr/local/projects/videautom/gateware/hw_validation/WbPlumbing/)
[info] Compiling 1 Scala source to /usr/local/projects/videautom/gateware/hw_validation/WbPlumbing/target/scala-2.12/test-classes ...
[error] /usr/local/projects/videautom/gateware/hw_validation/WbPlumbing/src/test/scala/wbplumbing/testwbplumbing.scala:26:5: Symbol 'type chisel3.experimental.MultiIOModule' is missing from the classpath.
[error] This symbol is required by 'type chisel3.tester.ChiselScalatestTester.T'.
[error] Make sure that type MultiIOModule is in your classpath and check for conflicting dependencies with `-Ylog-classpath`.
[error] A full rebuild may help if 'ChiselScalatestTester.class' was compiled against an incompatible version of chisel3.experimental.
[error]     test(new WbInterconPT(wbm, wbs)){ dut =>
[error]     ^
[error] /usr/local/projects/videautom/gateware/hw_validation/WbPlumbing/src/test/scala/wbplumbing/testwbplumbing.scala:26:10: type mismatch;
[error]  found   : wbplumbing.WbInterconPT
[error]  required: T
[error]     test(new WbInterconPT(wbm, wbs)){ dut =>
[error]          ^
[error] /usr/local/projects/videautom/gateware/hw_validation/WbPlumbing/src/test/scala/wbplumbing/testwbplumbing.scala:72:5: inferred type arguments [wbplumbing.WbInterconOneMaster] do not conform to method test's type parameter bounds [T <: chisel3.experimental.MultiIOModule]
[error]     test(new WbInterconOneMaster(wbm, Seq(wbs1, wbs2))) { dut =>
[error]     ^
[error] /usr/local/projects/videautom/gateware/hw_validation/WbPlumbing/src/test/scala/wbplumbing/testwbplumbing.scala:72:10: type mismatch;
[error]  found   : wbplumbing.WbInterconOneMaster
[error]  required: T
[error]     test(new WbInterconOneMaster(wbm, Seq(wbs1, wbs2))) { dut =>
[error]          ^
[error] /usr/local/projects/videautom/gateware/hw_validation/WbPlumbing/src/test/scala/wbplumbing/testwbplumbing.scala:75:7: not found: value step
[error]       step(1)
[error]       ^
[error] 5 errors found
[error] (Test / compileIncremental) Compilation failed
[error] Total time: 5 s, completed 6 nov. 2019 10:50:45
Makefile:7: recipe for target 'test' failed
```

Does somebody have a clue ?

edwardcwang, 2019-11-06:
Try publishing it locally? Does it work that way?

Martoni, 2019-11-06:
I have error publishing locally this project :
```
$ git clone https://github.com/ucb-bar/chisel-testers2.git
$ cd chisel-testers2
$ sbt publishLocal
[info] Loading settings for project chisel-testers2-build from plugins.sbt ...
[info] Loading project definition from /usr/local/opt/chisel-testers2/project
[info] Loading settings for project chisel-testers2 from build.sbt ...
[info] Set current project to chisel-testers2 (in build file:/usr/local/opt/chisel-testers2/)
[info] Wrote /usr/local/opt/chisel-testers2/target/scala-2.12/chisel-testers2_2.12-0.2-SNAPSHOT.pom
[info] Updating 
[info] Resolved  dependencies
[warn] 
[warn] 	Note: Unresolved dependencies path:
[error] sbt.librarymanagement.ResolveException: Error downloading edu.berkeley.cs:treadle_2.12:1.2-SNAPSHOT
[error]   Not found
[error]   Not found
[error]   not found: /home/fabien/.ivy2/local/edu.berkeley.cs/treadle_2.12/1.2-SNAPSHOT/ivys/ivy.xml
[error]   not found: https://repo1.maven.org/maven2/edu/berkeley/cs/treadle_2.12/1.2-SNAPSHOT/treadle_2.12-1.2-SNAPSHOT.pom
[error]   not found: https://oss.sonatype.org/content/repositories/snapshots/edu/berkeley/cs/treadle_2.12/1.2-SNAPSHOT/treadle_2.12-1.2-SNAPSHOT.pom
[error]   not found: https://oss.sonatype.org/content/repositories/releases/edu/berkeley/cs/treadle_2.12/1.2-SNAPSHOT/treadle_2.12-1.2-SNAPSHOT.pom
[error] Error downloading edu.berkeley.cs:chisel3_2.12:3.3-SNAPSHOT
[error]   Not found
[error]   Not found
[error]   not found: /home/fabien/.ivy2/local/edu.berkeley.cs/chisel3_2.12/3.3-SNAPSHOT/ivys/ivy.xml
[error]   not found: https://repo1.maven.org/maven2/edu/berkeley/cs/chisel3_2.12/3.3-SNAPSHOT/chisel3_2.12-3.3-SNAPSHOT.pom
[error]   not found: https://oss.sonatype.org/content/repositories/snapshots/edu/berkeley/cs/chisel3_2.12/3.3-SNAPSHOT/chisel3_2.12-3.3-SNAPSHOT.pom
[error]   not found: https://oss.sonatype.org/content/repositories/releases/edu/berkeley/cs/chisel3_2.12/3.3-SNAPSHOT/chisel3_2.12-3.3-SNAPSHOT.pom
[error] 	at lmcoursier.CoursierDependencyResolution.unresolvedWarningOrThrow(CoursierDependencyResolution.scala:245)
[error] 	at lmcoursier.CoursierDependencyResolution.$anonfun$update$34(CoursierDependencyResolution.scala:214)
[error] 	at scala.util.Either$LeftProjection.map(Either.scala:573)
[error] 	at lmcoursier.CoursierDependencyResolution.update(CoursierDependencyResolution.scala:214)
[error] 	at sbt.librarymanagement.DependencyResolution.update(DependencyResolution.scala:60)
[error] 	at sbt.internal.LibraryManagement$.resolve$1(LibraryManagement.scala:52)
[error] 	at sbt.internal.LibraryManagement$.$anonfun$cachedUpdate$12(LibraryManagement.scala:102)
[error] 	at sbt.util.Tracked$.$anonfun$lastOutput$1(Tracked.scala:69)
[error] 	at sbt.internal.LibraryManagement$.$anonfun$cachedUpdate$20(LibraryManagement.scala:115)
[error] 	at scala.util.control.Exception$Catch.apply(Exception.scala:228)
[error] 	at sbt.internal.LibraryManagement$.$anonfun$cachedUpdate$11(LibraryManagement.scala:115)
[error] 	at sbt.internal.LibraryManagement$.$anonfun$cachedUpdate$11$adapted(LibraryManagement.scala:96)
[error] 	at sbt.util.Tracked$.$anonfun$inputChanged$1(Tracked.scala:150)
[error] 	at sbt.internal.LibraryManagement$.cachedUpdate(LibraryManagement.scala:129)
[error] 	at sbt.Classpaths$.$anonfun$updateTask0$5(Defaults.scala:2942)
[error] 	at scala.Function1.$anonfun$compose$1(Function1.scala:49)
[error] 	at sbt.internal.util.$tilde$greater.$anonfun$$u2219$1(TypeFunctions.scala:62)
[error] 	at sbt.std.Transform$$anon$4.work(Transform.scala:67)
[error] 	at sbt.Execute.$anonfun$submit$2(Execute.scala:280)
[error] 	at sbt.internal.util.ErrorHandling$.wideConvert(ErrorHandling.scala:19)
[error] 	at sbt.Execute.work(Execute.scala:289)
[error] 	at sbt.Execute.$anonfun$submit$1(Execute.scala:280)
[error] 	at sbt.ConcurrentRestrictions$$anon$4.$anonfun$submitValid$1(ConcurrentRestrictions.scala:178)
[error] 	at sbt.CompletionService$$anon$2.call(CompletionService.scala:37)
[error] 	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
[error] 	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
[error] 	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
[error] 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
[error] 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
[error] 	at java.lang.Thread.run(Thread.java:748)
[error] (update) sbt.librarymanagement.ResolveException: Error downloading edu.berkeley.cs:treadle_2.12:1.2-SNAPSHOT
[error]   Not found
[error]   Not found
[error]   not found: /home/fabien/.ivy2/local/edu.berkeley.cs/treadle_2.12/1.2-SNAPSHOT/ivys/ivy.xml
[error]   not found: https://repo1.maven.org/maven2/edu/berkeley/cs/treadle_2.12/1.2-SNAPSHOT/treadle_2.12-1.2-SNAPSHOT.pom
[error]   not found: https://oss.sonatype.org/content/repositories/snapshots/edu/berkeley/cs/treadle_2.12/1.2-SNAPSHOT/treadle_2.12-1.2-SNAPSHOT.pom
[error]   not found: https://oss.sonatype.org/content/repositories/releases/edu/berkeley/cs/treadle_2.12/1.2-SNAPSHOT/treadle_2.12-1.2-SNAPSHOT.pom
[error] Error downloading edu.berkeley.cs:chisel3_2.12:3.3-SNAPSHOT
[error]   Not found
[error]   Not found
[error]   not found: /home/fabien/.ivy2/local/edu.berkeley.cs/chisel3_2.12/3.3-SNAPSHOT/ivys/ivy.xml
[error]   not found: https://repo1.maven.org/maven2/edu/berkeley/cs/chisel3_2.12/3.3-SNAPSHOT/chisel3_2.12-3.3-SNAPSHOT.pom
[error]   not found: https://oss.sonatype.org/content/repositories/snapshots/edu/berkeley/cs/chisel3_2.12/3.3-SNAPSHOT/chisel3_2.12-3.3-SNAPSHOT.pom
[error]   not found: https://oss.sonatype.org/content/repositories/releases/edu/berkeley/cs/chisel3_2.12/3.3-SNAPSHOT/chisel3_2.12-3.3-SNAPSHOT.pom
[error] Total time: 12 s, completed 6 nov. 2019 16:37:02

```

Martoni, 2019-11-06:
I suppose I have to publish Chisel3 and Treadle locally too ?

ucbjrl, 2019-11-06:
We expect to publish current SNAPSHOTs today. I'll try to remember to update this issue when they are available.

Martoni, 2019-11-06:
Ok I managed to publishLocal this project but it was a pain ! I had to publishLocal other projects in the right order before : 
- firrtl
- chisel3
- treadle
- chisel-testers2

Then I had to change version in my build.sbt : 
```scala
libraryDependencies += "edu.berkeley.cs" %% "chisel-testers2" % "0.2-SNAPSHOT"
...
val defaultVersions = Map(
  "chisel3" -> "3.3-SNAPSHOT",
...
```
And now it's working !

Martoni, 2019-11-06:
But I have to switch all my other project (spi2wb, MDIO, ...) with chisel3 3.3-SNAPSHOT .

ekiwi, 2021-08-17:
this should be fixed in the latest releases

closed 2021-08-17 by ekiwi