ucb-bar/chiseltest #74: Poking vecs of Bundles doesn't work

chick, 2019-10-14:
```scala
import chisel3._
import chisel3.tester._
import org.scalatest.FreeSpec

class VecIO extends Bundle {
  val x = UInt(5.W)
}

class UsesVec extends MultiIOModule {
  val in   = IO(Input(Vec(4, new VecIO)))
  val addr = IO(Input(UInt(8.W)))
  val out  = IO(Output(UInt(5.W)))

  out := in(addr)
}

class UsesVecSpec extends FreeSpec with ChiselScalatestTester {
  "run" in {
    test(new UsesVec) { c =>
      c.in(0).x.poke(5.U)
      c.in(1).x.poke(5.U)
      c.in(2).x.poke(4.U)
      c.addr.poke(2.U)
      c.clock.step()
      c.out.expect(4.U)
    }
  }
}
```
Generates error
```
Connection between sink (UInt<5>(IO in unelaborated UsesVec)) and source (VecIO(IO in unelaborated UsesVec)) failed @: Sink (UInt<5>(IO in unelaborated UsesVec)) and Source (VecIO(IO in unelaborated UsesVec)) have different types.
chisel3.internal.ChiselException: Connection between sink (UInt<5>(IO in unelaborated UsesVec)) and source (VecIO(IO in unelaborated UsesVec)) failed @: Sink (UInt<5>(IO in unelaborated UsesVec)) and Source (VecIO(IO in unelaborated UsesVec)) have different types.
```

ducky64, 2019-10-14:
Wait this looks like bad Chisel as opposed to a testers2 error. Isn't the error-ing line
```
out := in(addr)
```
where you're trying to assign a UInt from a Vec, and you don't even get into testers2?

chick, 2019-10-14:
Sorry, you are correct. I was bringing this over from user reported problem (unrelated to testers2) though I had elected to try and simulate it there. Looks like I messed up the conversion

closed 2019-10-14 by chick