ucb-bar/chiseltest #102: chisel-testers2 elaborates twice when Verilator is enabled

oharboe, 2019-12-17:
Elboration can be quite time consuming, so this slows down Verilator as a backend significantly.

Here's a pull request with a test that reproduces the problem and fails if it happens: https://github.com/ucb-bar/chisel-testers2/pull/98

Discussed here:

https://groups.google.com/forum/#!topic/chisel-users/31NHUTKm8gw

FIRRTL compile and Verilator only runs once though.

This test passes for Treadle, but fails for Verilator.

```
$ sbt "test:testOnly *SingleElaboration*"
[deleted]
[info] SingleElaborationTests:
[info] elaboration should happen once
[info] [0.004] Elaborating design...
[info] [0.124] Done elaborating.
[info] [0.000] Elaborating design...
[info] [0.009] Done elaborating.
HACK! new phases.WriteOutputAnnotations to save memory while waiting for Chisel 3.2.0 memory consumption fix
Total FIRRTL Compile Time: 631.7 ms
[deleted]
```

sequencer, 2020-04-28:
This should have been resolved.

oharboe, 2020-04-28:
@sequencer Which version?

oharboe, 2020-04-28:
@sequencer I made an automated test case #98  It should pass and some polished up version of it should be merged with master.

sequencer, 2020-04-28:
In the master? if not I think this should #156 should fix this.

sequencer, 2020-04-28:
iirc, #127 should have fixed this.

oharboe, 2020-04-28:
@sequencer  The only thing that is missing is an automated unit-test. I've rebased #98 on master, but master doesn't compile currently #160

carlosedp, 2022-03-21:
Maybe this doesn't apply anymore. Close?

oharboe, 2022-03-22:
This is working now, I don't know if there are any automated tests in master though.