ucb-bar/chiseltest #115: Wrap assertions in meaningful exception

ducky64, 2020-01-31:
This would require the backend interface layer to parse assertion failures from the simulation and return a meaningful exception. Ideally also reporting line numbers where the assertion failed and the message?

Might also be worth thinking about if the wrapping exception should be a chisel3 exception (and sharable among other projects) or if it should be specific to chiseltest.

edwardcwang, 2020-01-31:
Possibly related? #36

hcook, 2021-04-12:
It is also annoying/unfortunate that the treadle backed throws  `treadle.executable.StopException` while the Verilator backend throws `chiseltest.legacy.backends.verilator.TestApplicationException` --  this makes it harder to use scala-test's `assertThrows` method since I have to know what backend the user is going to select for a particular test (which ideally might be based on a specified test duration coming from the configuration)

hcook, 2021-04-12:
Also the fact that `TestApplicationException` is `private` makes it uh... hard to catch ... and do my own thing with
```
catch e: Exception if e.getClass.getName == "chiseltest.legacy.backends.verilator.TestApplicationException" =>
```
😂

ekiwi, 2021-04-12:
> Also the fact that `TestApplicationException` is `private` makes it uh... hard to catch ... and do my own thing with

I agree. This needs a backend independent API. Do you think we should separate `Stop` statements firing from assertion violations (using our new `assert` or `assume` statement support)?
Should `Stop` actually lead to an exception, or should it just return the return code from the `test` invocation?

hcook, 2021-04-13:
> Do you think we should separate Stop statements firing from assertion violations (using our new assert or assume statement support)?

As long as I get back a public case class containing all this information I am happy for it to be a single one that handles everything that happened during a simulation, but breaking it down into stop/assert/assume cases sounds reasonable too. https://github.com/ucb-bar/chisel-testers2/issues/173 also seems relevant in terms of considering valid simulation-ending events that might involve multiple assertions firing. And per https://github.com/ucb-bar/chisel-testers2/issues/243,  "Timing out" of a max cycle count likewise should be a first-class citizen among these cases, though personally I had an easy enough time dealing with that already via `dut.clock.setTimeout` + `assertThrows[TimeoutException]`

> Should Stop actually lead to an exception, or should it just return the return code from the test invocation?

Just giving back the the return code makes sense to me, but I don't actually know/ys

ekiwi, 2021-10-21:
this should now be fixed by: https://github.com/ucb-bar/chisel-testers2/pull/377

closed 2021-10-21 by ekiwi