ucb-bar/chiseltest #129: Maven repo not found

nothinn, 2020-02-21:
I have an issue where I have tried adding the tester as described in the readme. However, when running the test, it claims that the maven repo is not found and when I look in the maven repo, it is not there either:

```shell
[error] (lib / update) sbt.librarymanagement.ResolveException: Error downloading edu.berkeley.cs:chiseltest_2.12:0.1-SNAPSHOT
[error]   Not found
[error]   Not found
[error]   not found: /home/simon/.ivy2/local/edu.berkeley.cs/chiseltest_2.12/0.1-SNAPSHOT/ivys/ivy.xml
[error]   not found: https://repo1.maven.org/maven2/edu/berkeley/cs/chiseltest_2.12/0.1-SNAPSHOT/chiseltest_2.12-0.1-SNAPSHOT.pom
[error]   not found: https://oss.sonatype.org/content/repositories/snapshots/edu/berkeley/cs/chiseltest_2.12/0.1-SNAPSHOT/chiseltest_2.12-0.1-SNAPSHOT.pom
[error]   not found: https://oss.sonatype.org/content/repositories/releases/edu/berkeley/cs/chiseltest_2.12/0.1-SNAPSHOT/chiseltest_2.12-0.1-SNAPSHOT.pom
[error] Total time: 5 s, completed Feb 21, 2020 3:14:09 PM
```

This is with the following in my build.sbt:
```sbt
 libraryDependencies ++= Seq("edu.berkeley.cs" %% "chiseltest" % "0.1-SNAPSHOT")
```

When changing the imports to *chisel3.tester.* instead of *chiseltest.*  and using
``` sbt
libraryDependencies ++= Seq("edu.berkeley.cs" %% "chisel-testers2" % "0.1-SNAPSHOT")
```
in the build.sbt file, then it works fine. I am using *chisel3 3.3-SNAPSHOT* with it.

edwardcwang, 2020-02-21:
Hello Simon, thank you for your interest in Chisel and ChiselTest! Could you please try with the following line in `build.sbt` and let us know if it works?

```
libraryDependencies ++= Seq("edu.berkeley.cs" %% "chiseltest" % "0.2-SNAPSHOT")
```

nothinn, 2020-02-22:
That worked perfectly. Now I can do concurrent testing on my SoC!

Thank you. (Remember to update the Readme for other users)

closed 2020-02-22 by edwardcwang

schoeberl, 2020-02-23:
If you want to avoid the need of an Internet connection every time you build your project, you could use the libraries with a version and avoiding SNAPSHOT. For the tester the latest version is:

libraryDependencies += "edu.berkeley.cs" %% "chisel-testers2" % "0.1.2"

Looks like we are waiting for publishing of a 0.2.x as chiseltest on Maven.

Martin

PR #130 (2020-02-22): Update chiseltest version in README
Close #129