ucb-bar/chiseltest #141: java.lang.NoClassDefFoundError: firrtl/Flow

ekiwi, 2020-03-20:
My `build.sbt` library dependencies:
```
libraryDependencies += "edu.berkeley.cs" %% "chisel3" % "3.2.+"
libraryDependencies += "edu.berkeley.cs" %% "chisel-testers2" % "0.1.+" % Test
```

My Test:
```
class SplitSpec extends FlatSpec with ChiselScalatestTester with Matchers {
  it should "split bits correctly" in {
    test(new SplitTestModule(width = 4, splitAt = 3)) { dut =>
      dut.io.in.poke("b1000".U)
      dut.io.msb.expect("b1".U)
      dut.io.lsb.expect("b000".U)
      dut.io.concat_msb_lsb.expect("b1000".U)
    }
  }
}
```

When I try to run the test (this is my only test) in `sbt` I get:
```
[info] [0.005] Elaborating design...
[info] [0.204] Done elaborating.
Total FIRRTL Compile Time: 564.7 ms
[info] SplitSpec:
[info] utils.SplitSpec *** ABORTED ***
[info]   java.lang.NoClassDefFoundError: firrtl/Flow
[info]   at treadle.stage.phases.PrepareAst$.<init>(PrepareAst.scala:117)
[info]   at treadle.stage.phases.PrepareAst$.<clinit>(PrepareAst.scala)
[info]   at treadle.stage.TreadleTesterPhase$.<init>(TreadleStage.scala:46)
[info]   at treadle.stage.TreadleTesterPhase$.<clinit>(TreadleStage.scala)
[info]   at chisel3.tester.backends.treadle.TreadleExecutive$.start(TreadleExecutive.scala:47)
[info]   at chisel3.tester.defaults.package$.createDefaultTester(defaults.scala:24)
[info]   at chisel3.tester.ChiselScalatestTester$TestBuilder.apply(ChiselScalatestTester.scala:26)
[info]   at utils.SplitSpec.$anonfun$new$1(SplitSpec.scala:25)
[info]   at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:12)
[info]   at org.scalatest.OutcomeOf.outcomeOf(OutcomeOf.scala:85)
[info]   ...
[info]   Cause: java.lang.ClassNotFoundException: firrtl.Flow
[info]   at java.net.URLClassLoader.findClass(URLClassLoader.java:382)
[info]   at sbt.internal.ManagedClassLoader.findClass(ManagedClassLoader.java:98)
[info]   at java.lang.ClassLoader.loadClass(ClassLoader.java:419)
[info]   at java.lang.ClassLoader.loadClass(ClassLoader.java:352)
[info]   at treadle.stage.phases.PrepareAst$.<init>(PrepareAst.scala:117)
[info]   at treadle.stage.phases.PrepareAst$.<clinit>(PrepareAst.scala)
[info]   at treadle.stage.TreadleTesterPhase$.<init>(TreadleStage.scala:46)
[info]   at treadle.stage.TreadleTesterPhase$.<clinit>(TreadleStage.scala)
[info]   at chisel3.tester.backends.treadle.TreadleExecutive$.start(TreadleExecutive.scala:47)
[info]   at chisel3.tester.defaults.package$.createDefaultTester(defaults.scala:24)
```

Am I doing something wrong or is this a bug? Do I have to explicitly add `firrtl` to my dependencies?

ekiwi, 2020-03-20:
So adding `firrtl` and `treadle` explicitly to my dependencies does not seem to help:
```
libraryDependencies += "edu.berkeley.cs" %% "firrtl" % "1.2.+"
libraryDependencies += "edu.berkeley.cs" %% "treadle" % "1.1.+"
```

Same error.

edwardcwang, 2020-03-20:
I've been using `chiseltest` `0.2-SNAPSHOT` with Chisel 3.2.x and it seems to work.

ucbjrl, 2020-03-20:
I think `Flow` is a feature of FIRRTL 1.3. You'll need to use chisel3 `3.3-SNAPSHOT`s, chiseltest (or chisel-testers2) `0.2-SNAPSHOT`

ekiwi, 2020-03-20:
Does that mean that the version published to maven isn't compatible with the rest of the chisel ecosystem that is published there?
[chisel-testers2](https://mvnrepository.com/artifact/edu.berkeley.cs/chisel-testers2) was updated in March 2020.

ucbjrl, 2020-03-20:
We publish two series of versions to Maven. Each series contains compatible versions. The two versions are:
- the _current_ stable release series: reported as `latest release` in the Releases tab of each repository,
- the _current_ SNAPSHOT series: as close to master as possible, destined to be the next major release.

Each series contains compatible versions of each project in the chisel ecosystem.

ekiwi, 2020-03-20:
OK. So I got it to work by specifying all the versions explicitly.

```
libraryDependencies += "edu.berkeley.cs" %% "chisel3" % "3.2.4"
libraryDependencies += "edu.berkeley.cs" %% "chisel-testers2" % "0.1.4" % Test
```

Thanks!

closed 2020-03-20 by ekiwi

ucbjrl, 2020-03-20:
Your initial dependencies should have worked, that is, I would have expected 3.2.+ to fetch 3.2.4 and 0.1.+ to fetch 0.1.4. I'll try to investigate.