ucb-bar/chiseltest #147: asTypeOf equivalence to cast between types

sequencer, 2020-04-02:
Currently I'm using tester2 to test diplomatic modules.
I already write a firrtl transform to boring `auto` IO of diplomatic fuzzer module up to top IO,
e.g.
```
    val monitor: AutoBundle = getIO(fuzzer.module.auto, "monitor")
```
If I access this monitor, I can only get `AutoBundle`, 
However, in tester2, `asTypeOf` is not allowed, so the question is: "how can I cast `AutoBundle` to `TLBundle` for peek poke."
Here is my currently test
https://github.com/sequencer/diplomatictester/tree/master/tests/src/TopWireSpec.scala

ducky64, 2020-04-02:
The underlying problem is that `asTypeOf` is evaluated at runtime (simulation time), not compile-time, so the testbench doesn't know the resulting value (only that it's a node that is the result of a type-cast). It's the same reason why you can't do something like `x.poke(1.U + 2.U)`, since the result of `1.U + 2.U` isn't (currently) known at elaboration time.

The idea of elaboration-time const-prop has been brought up in chisel development discussions, though there's a concern of subtle implementation differences between what FIRRTL does and bugs if we re-implemented it in chisel frontend. There was also the idea of making a FIRRTL const-prop library that could be invoked by the frontend, though I don't think it's been priority enough to have gone anywhere.

In the meantime, you can write a helper method that does a literal-to-literal (literal extraction to Scala-land value, followed by literal constructor from those values to the new type) that happens without using asTypeOf. It's less clean than `asTypeOf`, but it's a stopgap solution?

sequencer, 2020-04-03:
Thank you! Currently I'm trying to use edge to reconstruct literal TileLink `AutoBundle` in tester2, and poke them into `AutoBundle` IO.
However I found there seems to be no `Lit` method for `AutoBundle`, since it is generated from `Record`, so I may need to implement a `Lit` function like it does in `Bundle`?
Is this a rational approach to implement this?

sequencer, 2020-04-03:
And there is another question:
is possible to poke a Duplex IO like with a Bundle LitValue?
```
Bundle {
  val in = Input(Bool())
  val out = Output(Bool())
}
```

ducky64, 2020-04-03:
That sounds reasonable, though because Record is a more advanced API, support for it might be patchy. You might see if you can change Lit to directly work on Record, otherwise you can write your own. [The implementation of Lit](https://github.com/freechipsproject/chisel3/blob/master/core/src/main/scala/chisel3/experimental/package.scala#L132) might be a useful resource. You might also have to update the poke definition in tester2 to take a Record, since it's currently limited to a Bundle.

In theory, this seems architecturally clean and consistent if you want to try it (but as always, the devil is in the details). If it works, we take PRs :)

closed 2020-04-08 by sequencer