ucb-bar/chiseltest #148: Extraction ScalaBlackBox from treadle as a common BlackBox implmentation.

sequencer, 2020-04-03:
In treadle, we have `ScalaBlackBox` for circuit behavior model. However this behavior model can only be executed by treadle but no verilator/vcs. In the verilog verification, We cannot use treadle.
But I think it's possible to chisel-tester2 to connect treadle backend between vcs/verilator backend for co-simulation. we can use treadle to instantiate different treadle instance, which will execute different  `ScalaBlackBox` only, then we can use tester2 as a bridge to glue verilog simulation with treadle behavior model together for cosimulation.

ducky64, 2020-04-03:
In theory, maybe. In practice, it will probably be super tricky to plumb everything together, especially using components in ways they weren't designed to be used - so maybe the better solution is to make a dedicated / standalone cosimulation wrapper.

There was some discussion a while back about testers2 not being the simulation top-level, though that isn't the case right now - and requires significant thought into design to make that happen.

ekiwi, 2023-01-11:
Adding a BlackBox like this to Verilator is difficult to impossible and I do not see such a big value in this. Thus this most likely will never be implemented.

closed 2023-01-11 by ekiwi