ucb-bar/chiseltest #154: Only log limited number of signals to VCD

nothinn, 2020-04-16:
Feature request:

* When running large systems for an extended period of time, the generated VCD files become very large. I can see that Verilator has support for only logging selected signals such as by setting the depth of components to log or log signals starting with underscore _, only by changing the *--trace* flag to *--trace-depth levels* or *--trace-underscore*.
* * Is there an easy way to hack this for my own testing?

* Verilator also supports to start logging at a later time. This would be very handy as it would be possible to skip a lot of initialization or only start logging when an error is found, but this is not as useful and more difficult to implement.

I don't know how easy this is to implement in Treadle, but as this is primarily a concern for large circuits, Verilator is probably the best point to start with.

ducky64, 2020-04-17:
Is this something that can be accomplished with a mechanism to propagate custom command line flags to Verilator? Or does it need some deeper functionality, like generated Verilog modifications?

nothinn, 2020-04-17:
For only tracing to a certain depth, it can easily be done by only propagating the "--trace-depth n" flag where n is the number of levels. This will be very helpful at first, as signals can always be routed out to the top level for debugging if needed.

For tracing signals starting with an underscore, the generated Verilog would need to emit signal names starting with an underscore when being marked as *log*. This could be in the same way as signals are marked *dontTouch*. I don't know how the inner mechanics should handle this. I am not quite sure whether --trace-underscore will add signals that start with an underscore or only trace signals that start with an underscore.

For starting the logging at a later point in time, this seems a bit more complicated. It is something about waiting to call *VerilatedVcdC->open* until it is actually needed.

TL;DR: Propagating custom command line flags to Verilator should work.

sequencer, 2020-04-17:
I think Tester2 already provided `VerilatorCFlags` and `VerilatorFlags` fot this purpose.

nothinn, 2020-04-17:
Is this done by setting an annotation like *VerilatorBackendAnnotation* and *WriteVcdAnnotation*? If not, do you know how to do it without hacking the project? 

One solution I see is to copy the project and modify [this line](https://github.com/ucb-bar/chisel-testers2/blob/14ee517144e9a3def8a0a04941d8e662a8db6bda/src/main/scala/chiseltest/legacy/backends/verilator/VerilatorExecutive.scala#L79) which I believe is the one that sets the flag.

ducky64, 2020-04-21:
Does passing in an annotation like [this test case](https://github.com/ucb-bar/chisel-testers2/blob/master/src/test/scala/chiseltest/tests/OptionsPassingTest.scala#L25) work? For example,
```scala
test(new MyModule())
  .withAnnotations(Seq(
      VerilatorFlags(Seq("--flag1", "--flag2"))
  )) { c =>
...
}
```

nothinn, 2020-04-22:
Yeah, this works perfectly in conjunction with WriteVcdAnnotation (WriteVcdAnnotation is still needed):

`.withAnnotations(Seq(VerilatorBackendAnnotation, WriteVcdAnnotation, VerilatorFlags(Seq("--trace-depth 2", "--threads 12"))))`

(Also found out that Verilator has multi-threading support)

Thank you, now I can get on with testing large systems over an extended time!

closed 2020-04-25 by ducky64