ucb-bar/chiseltest #158: Allow expect with BitPat.

sequencer, 2020-04-25:
In the bus protocol expecting.
There is a case that we only need to expecting a part of signals.
For example:
To test TileLink protocol, I'm expecting a 128-bit width of data with mask in `0b00000000000000000000000011111111`, I expect data to be `0h12345678`. but dut return a `0h12345678123456781234567812345678`. other than `12345678` in LSB, all data should be ignored  in expecting. So `BitPat` as literal, UInt/Bits as Data should be implemented for this case.

ducky64, 2020-04-25:
Yeah, I think it makes sense to create a testableBits class that can take a BitPat as the comparison value, if you want to write an PR it. That being said, though BitPat is a Chisel construct, the implementation is more bolted on in utils, and usage / support probably isn't as uniform as other core Chisel constructs.

sequencer, 2020-04-26:
Yes I'm going to PR it after I finish #156

nullobject, 2020-05-29:
Being able to write `io.foo.expect("b????_???1".U)` in my tests would be amazing. :heart: 

Is this what you're suggesting?

ducky64, 2020-06-04:
Yes, I think that is the idea.

Nit: the syntax might be more of `io.foo.expect(BitPat("b????_???1"))`, since UInts don't support unknown fields, so BitPat is a separate class entirely.

kammoh, 2021-07-16:
Hi @sequencer!
Any updates on this feature?