ucb-bar/chiseltest #160: Compilation error on master after PR #151 merged

oharboe, 2020-04-28:
@sequencer I'm having problems compiling the master branch after #151 merged:

```
$ sbt "test:testOnly *SingleElabor*"
[deleted]
[error] .../chisel-testers2/src/main/scala/chiseltest/package.scala:25:26: value checkTypeEquivalence is not a member of object chisel3.experimental.DataMirror
[error]  Note: implicit method validToDriver is not applicable here because it comes after the application point and it lacks an explicit result type
[error]       require(DataMirror.checkTypeEquivalence(x, value), s"Record type mismatch")
[error]                          ^
[error] .../chisel-testers2/src/main/scala/chiseltest/package.scala:45:26: value checkTypeEquivalence is not a member of object chisel3.experimental.DataMirror
[more of the same deleted]
```

sequencer, 2020-04-28:
Thanks for testing this, you should build with chisel3 in master branch.

oharboe, 2020-04-28:
@sequencer Ah, I don't have that in my skillset yet for messing about with chisel-tester2 locally...

oharboe, 2020-04-28:
@sequencer  Pull requests fail to build with latest chisel-tester2 master, so this problem affects more than just a local workflow.  Look at #159 it only adds a .gitignore file, it now fails...

sequencer, 2020-04-28:
You may need to clone chisel3 firstly, and use `sbt publishLocal` to install that SNAPSHOT to your local ivy repo.

sequencer, 2020-04-28:
Try to add in your PR
```
### Build Dependencies
Build with: git-clone firrtl https://github.com/freechipsproject/firrtl.git master
Build with: git-clone chisel3 https://github.com/freechipsproject/chisel3.git master
```

ucbjrl, 2020-04-28:
Add the lines suggested by @sequencer to the PR body (not as a comment). Then you'll need to push a trivial change to re-trigger the test build.

oharboe, 2020-04-28:
@ucbjrl @sequencer PR's updated... Fingers x'ed!

oharboe, 2020-04-28:
Thanks for explaining how this workflow works. I gather that this procedure is not normally needed and things just work out of the box with master, given that it was only the latest commit on master that needed this special PR treatment.

For me this issue is documented and solved.

closed 2020-04-28 by oharboe