ucb-bar/chiseltest #165: library for testing standard components like wishbone, uart, ...

Martoni, 2020-05-27:
Is there some examples or standard libraries for testing components like wishbone, uart, spi, ... with chisel-testers2 ?

ducky64, 2020-05-27:
I've heard of people playing with this for various protocols, but I'm not aware of any released or example/documentation-grade code. One potential issue with a testing library is that it needs a underlying common bundle definition. The Decoupled drivers are provided here because it's defined in chisel3 base, but wishbone, UART, and SPI aren't.

Here an [example SPI (slave side) driver code](https://gist.github.com/ducky64/4c004097ff559c9334bed89b518a29a5) I wrote as part of a FPGA project.

Martoni, 2020-05-28:
Writing the test bench is the most time and resource consuming part of the development process. If we could get some basic bricks to ease and speed up development it would make Chisel3 even more attractive I think.
Personnaly I use the [CocoTB](https://docs.cocotb.org/en/latest/) python library a lot because I find it easier than chisel tester even with Chisel designs.

ducky64, 2020-06-04:
I agree that this reusable library ultimately this is the dream and goal of chiseltest, but also realistically, there's a lot of groundwork in standardizing chisel libraries that are necessary before this can happen.

That being said, this issue thread might be a good starting point to think about what protocols (especially for Bundles defined in common, including community, libraries) could use drivers. In the short term, I think a more realistic approach would be supporting adding testdrivers to those libraries. (except for anything in chisel base, including `chisel3.utils`, for which drivers should be provided in this repo)

schoeberl, 2020-06-05:
This is probably a lot of work and good to start the discussion here. Having testing infrastructure, like common bus protocol transactions available, is actually a strong argument for UVM. That is what we are competing with, right?

Cheers,
Martin