ucb-bar/chiseltest #172: Support Blackbox and RawModule

sequencer, 2020-06-18:
Since tester2 only has cycle-based simulator interface, async multi-clock seems not be possible. 
But main clock can be supported, after #156, we can set clock by annotation, and let user define arbitrary reset logic to remove dependency on chisel implicit clock/reset.

ducky64, 2020-06-19:
Related: #14, a more general case of driving clock signals within the testbench and supporting true multiclock designs. But, if your use case doesn't need the full power of multiple independent clocks (which will take a lot longer - since it's a complex feature that hasn't even been fully reasoned through yet) and is just a rename, this makes sense. What's the example use case?

sequencer, 2020-06-19:
Yes, basically, it's a rename. I think the most important use case is directly test a verilog or other `RawModule` without wrapping it.
And another case is that if user play with clock domain, init reset logic should be changed(maybe multiple reset, or reset more than one cycle), than hardcoded logic should be changed too.