ucb-bar/chiseltest #173: throw multi exceptions

sequencer, 2020-06-18:
I notice there will be a case that multiple assertion will be raised in one cycle, do I need to collect them together and throw them, but I notice scalatest and utest only support one sourceline trace back. @ducky64 what's your idea to this?

ducky64, 2020-06-19:
Yeah, so the issue here is that imperative programs typically run one line at a time (and tests fail immediately), whereas that same abstraction makes less sense in hardware where things are naturally parallel.

That being said, I think there's a few different cases for multi-exceptions:
- expect failures, in a single thread: I don't think it makes sense to throw multiple exceptions here, since single-thread semantics in a testdriver should match imperative semantics.
- expect failures, across threads: less clear what should happen here - if a thread has an exception, should it stop simulation immediately, or finish running the time step? One thing to note is that chiseltest actually specifies a deterministic thread ordering (threads are linearized, in a sense), so maybe imperative semantics should apply even across threads. But on the other hand, it is conceptually parallel.
- assertion failures: I think there's a clear argument here for displaying all the assertions simultaneously (assuming the underlying simulator supports that)

In general, I think this should be use-case driven. Is there a good example design, where it would have been helpful while debugging to see multiple expect failures or hardware assertions that failed simultaneously?

If we do want to implement this, the jankiest option is just to ignore ScalaTest's exception source locator field, and dump all the source locators in the exception message body. But how to usefully display a bunch of exceptions (within the constraints of a console - as opposed to richer text formats that support expansion or linking) is also a question.

sequencer, 2020-06-19:
Currently, my implementation is generating `Exception`, put them into an buffer, for each cycle, if the element inside this buffer is non-empty, simulator will stop. and let ScalaTest collect this annotation and throw it.
It is possible to use the stack trace to get the source locator and put them in the message, but `TestFailedException` can only have 1 `failedCodeStackDepth`, so I just put the source locator of first exceptions to which.

closed 2023-06-27 by ekiwi