ucb-bar/chiseltest #175: What is the proper treadle version for the chisel-testers2?

HaFred, 2020-06-25:
I am trying to append the chisel-testers2 in [chipyard-1.3.0](https://github.com/ucb-bar/chipyard/tree/1.3.0), however the treadle version of chipyard seems not matched with chisel-testers2 because the tests are not functioning at all.

chick, 2020-06-26:
I think it is probably ce521f42488c209e876f5b0b7810cfcdd84d8348
But it's a little complicated to determine because chipyard seems to be using SNAPSHOTS versions
for its submodules.
The above seems to work for me, can you give it a try

HaFred, 2020-06-26:
> I think it is probably [ce521f4](https://github.com/ucb-bar/chisel-testers2/commit/ce521f42488c209e876f5b0b7810cfcdd84d8348)
> But it's a little complicated to determine because chipyard seems to be using SNAPSHOTS versions
> for its submodules.
> The above seems to work for me, can you give it a try

Hi chick, thank you for your reply. Sorry that I don't get the treadle version I should update in my chipyard fork to make treadle working by the https://github.com/ucb-bar/chisel-testers2/commit/ce521f42488c209e876f5b0b7810cfcdd84d8348, could you elaborate on this a little bit? Thx

HaFred, 2020-06-26:
Or you meant I could try https://github.com/ucb-bar/chisel-testers2/commit/ce521f42488c209e876f5b0b7810cfcdd84d8348 this chisel-testers2 in the chipyard 1.3.0. If so, I will do and get back to you with the result

HaFred, 2020-06-26:
I found out I have to `git checkout 8bf6be704e44cc5edc3a1b41aae35ab4c0bbbfab` under `chipyard/tools/treadle` to make the chiselTests working, even though I cannot access where this commit should be (https://github.com/freechipsproject/commit/8bf6be704e44cc5edc3a1b41aae35ab4c0bbbfab). And the chiselTests' commit version seems not affecting the test, the treadle's does.

Regards,

HaFred, 2020-06-27:
Both the testers2 & iotester are not the suggested testing tools in chipyard envs, it is understandable there are dependencies issues to disturb these 2 tools.

closed 2020-06-27 by HaFred

chick, 2020-06-28:
@HaFred Nice job figuring that out. I'm adding looking into getting this straightened out to my to do list. Thanks