ucb-bar/chiseltest #177: Verilated::flushCall renamed in newer versions of verilator

jiegec, 2020-07-20:
Code change in verilator: https://github.com/verilator/verilator/commit/fac89c5d6233f900d0e382bde21526400427e388#diff-6c80b812607617c162e46fd56e6579c8

Related code in chisel-testers2:
https://github.com/ucb-bar/chisel-testers2/blob/db85e6fcb5ce2d8397ea0d9ed1745cdd5b7b9e35/src/main/scala/chiseltest/legacy/backends/verilator/VerilatorCppHarnessGenerator.scala#L148

ljwljwljwljw, 2020-07-31:
I have same problem, maybe "Verilated::flushCall()" need to be changed to "Verilated::runFlushCallbacks()"?

ducky64, 2020-08-07:
I think it makes sense for us to keep our Verilator version in sync with the [version recommended by chisel3's setup](https://github.com/freechipsproject/chisel3/blob/master/SETUP.md), which is currently 4.016.

Though, feel free to make a PR if you have code that updates behavior for newer versions while remaining 4.016 compatibility?

ljwljwljwljw, 2020-08-08:
> I think it makes sense for us to keep our Verilator version in sync with the [version recommended by chisel3's setup](https://github.com/freechipsproject/chisel3/blob/master/SETUP.md), which is [...]

Thanks for you advice, here(#187) is my PR, I just simply use `verilator --version` to check verilator's version as a temporary solution, I noticed that in #156 we can use our own cpp harness or `SimulatorBinary`, after that PR this problem will be well solved.

closed 2020-08-08 by ducky64