ucb-bar/chiseltest #178: Both the verilator & treadle backend throw exceptions

HaFred, 2020-07-24:
Hi @sequencer, after I bumped the latest commit, these 2 backends throw the same exception
```
[info] [0.004] Elaborating design...
[info] [1.922] Done elaborating.
Computed transform order in: 273.8 ms
Total FIRRTL Compile Time: 2223.3 ms

An exception or error caused a run to abort: org.json4s.FullTypeHints.<init>(Lscala/collection/immutable/List;)V 
java.lang.NoSuchMethodError: org.json4s.FullTypeHints.<init>(Lscala/collection/immutable/List;)V
	at firrtl.annotations.JsonProtocol$.jsonFormat(JsonProtocol.scala:132)
	at firrtl.annotations.JsonProtocol$.serializeTry(JsonProtocol.scala:152)
	at firrtl.annotations.JsonProtocol$.serialize(JsonProtocol.scala:144)
	at firrtl.options.phases.WriteOutputAnnotations.transform(WriteOutputAnnotations.scala:38)
	at firrtl.options.phases.WriteOutputAnnotations.transform(WriteOutputAnnotations.scala:15)
	at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:37)
	at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:15)
	at firrtl.options.Translator.transform(Phase.scala:240)
	at firrtl.options.Translator.transform$(Phase.scala:240)
	at firrtl.options.phases.DeletedWrapper.transform(DeletedWrapper.scala:15)
	at firrtl.options.Stage.$anonfun$transform$5(Stage.scala:46)
	at scala.collection.LinearSeqOptimized.foldLeft(LinearSeqOptimized.scala:126)
	at scala.collection.LinearSeqOptimized.foldLeft$(LinearSeqOptimized.scala:122)
	at scala.collection.immutable.List.foldLeft(List.scala:89)
	at firrtl.options.Stage.$anonfun$transform$3(Stage.scala:46)
	at logger.Logger$.$anonfun$makeScope$2(Logger.scala:168)
	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:62)
	at logger.Logger$.makeScope(Logger.scala:166)
	at firrtl.options.Stage.transform(Stage.scala:46)
	at chisel3.stage.phases.MaybeFirrtlStage.$anonfun$transform$2(MaybeFirrtlStage.scala:19)
	at scala.Option.getOrElse(Option.scala:189)
	at chisel3.stage.phases.MaybeFirrtlStage.transform(MaybeFirrtlStage.scala:19)
	at chisel3.stage.phases.MaybeFirrtlStage.transform(MaybeFirrtlStage.scala:13)
	at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:37)
	at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:15)
	at firrtl.options.Translator.transform(Phase.scala:240)
	at firrtl.options.Translator.transform$(Phase.scala:240)
	at firrtl.options.phases.DeletedWrapper.transform(DeletedWrapper.scala:15)
	at firrtl.options.DependencyManager.$anonfun$transform$3(DependencyManager.scala:262)
	at scala.collection.LinearSeqOptimized.foldLeft(LinearSeqOptimized.scala:126)
	at scala.collection.LinearSeqOptimized.foldLeft$(LinearSeqOptimized.scala:122)
	at scala.collection.immutable.List.foldLeft(List.scala:89)
	at firrtl.options.DependencyManager.transform(DependencyManager.scala:255)
	at firrtl.options.DependencyManager.transform$(DependencyManager.scala:240)
	at firrtl.options.PhaseManager.transform(DependencyManager.scala:419)
	at chisel3.stage.ChiselStage.run(ChiselStage.scala:35)
	at chiseltest.legacy.backends.verilator.VerilatorExecutive$.start(VerilatorExecutive.scala:61)
	at chiseltest.defaults.package$.createDefaultTester(defaults.scala:24)
	at chiseltest.ChiselScalatestTester$TestBuilder.apply(ChiselScalatestTester.scala:30)
	at dla.tests.petest.ProcessingElementSpecTest.$anonfun$new$11(ProcessingElementSpecTest.scala:674)
	at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:23)
	at org.scalatest.OutcomeOf.outcomeOf(OutcomeOf.scala:85)
	at org.scalatest.OutcomeOf.outcomeOf$(OutcomeOf.scala:83)
	at org.scalatest.OutcomeOf$.outcomeOf(OutcomeOf.scala:104)
	at org.scalatest.Transformer.apply(Transformer.scala:22)
	at org.scalatest.Transformer.apply(Transformer.scala:20)
	at org.scalatest.FlatSpecLike$$anon$5.apply(FlatSpecLike.scala:1682)
	at org.scalatest.TestSuite.withFixture(TestSuite.scala:196)
	at org.scalatest.TestSuite.withFixture$(TestSuite.scala:195)
	at dla.tests.petest.ProcessingElementSpecTest.chiseltest$ChiselScalatestTester$$super$withFixture(ProcessingElementSpecTest.scala:488)
	at chiseltest.ChiselScalatestTester.$anonfun$withFixture$1(ChiselScalatestTester.scala:47)
	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:62)
	at chiseltest.ChiselScalatestTester.withFixture(ChiselScalatestTester.scala:47)
	at chiseltest.ChiselScalatestTester.withFixture$(ChiselScalatestTester.scala:44)
	at dla.tests.petest.ProcessingElementSpecTest.withFixture(ProcessingElementSpecTest.scala:488)
	at org.scalatest.FlatSpecLike.invokeWithFixture$1(FlatSpecLike.scala:1680)
	at org.scalatest.FlatSpecLike.$anonfun$runTest$1(FlatSpecLike.scala:1692)
	at org.scalatest.SuperEngine.runTestImpl(Engine.scala:286)
	at org.scalatest.FlatSpecLike.runTest(FlatSpecLike.scala:1692)
	at org.scalatest.FlatSpecLike.runTest$(FlatSpecLike.scala:1674)
	at org.scalatest.FlatSpec.runTest(FlatSpec.scala:1685)
	at org.scalatest.FlatSpecLike.$anonfun$runTests$1(FlatSpecLike.scala:1750)
	at org.scalatest.SuperEngine.$anonfun$runTestsInBranch$1(Engine.scala:393)
	at scala.collection.immutable.List.foreach(List.scala:392)
	at org.scalatest.SuperEngine.traverseSubNodes$1(Engine.scala:381)
	at org.scalatest.SuperEngine.runTestsInBranch(Engine.scala:370)
	at org.scalatest.SuperEngine.$anonfun$runTestsInBranch$1(Engine.scala:407)
	at scala.collection.immutable.List.foreach(List.scala:392)
	at org.scalatest.SuperEngine.traverseSubNodes$1(Engine.scala:381)
	at org.scalatest.SuperEngine.runTestsInBranch(Engine.scala:376)
	at org.scalatest.SuperEngine.runTestsImpl(Engine.scala:458)
	at org.scalatest.FlatSpecLike.runTests(FlatSpecLike.scala:1750)
	at org.scalatest.FlatSpecLike.runTests$(FlatSpecLike.scala:1749)
	at org.scalatest.FlatSpec.runTests(FlatSpec.scala:1685)
	at org.scalatest.Suite.run(Suite.scala:1124)
	at org.scalatest.Suite.run$(Suite.scala:1106)
	at org.scalatest.FlatSpec.org$scalatest$FlatSpecLike$$super$run(FlatSpec.scala:1685)
	at org.scalatest.FlatSpecLike.$anonfun$run$1(FlatSpecLike.scala:1795)
	at org.scalatest.SuperEngine.runImpl(Engine.scala:518)
	at org.scalatest.FlatSpecLike.run(FlatSpecLike.scala:1795)
	at org.scalatest.FlatSpecLike.run$(FlatSpecLike.scala:1793)
	at org.scalatest.FlatSpec.run(FlatSpec.scala:1685)
	at org.scalatest.tools.SuiteRunner.run(SuiteRunner.scala:45)
	at org.scalatest.tools.Runner$.$anonfun$doRunRunRunDaDoRunRun$13(Runner.scala:1349)
	at org.scalatest.tools.Runner$.$anonfun$doRunRunRunDaDoRunRun$13$adapted(Runner.scala:1343)
	at scala.collection.immutable.List.foreach(List.scala:392)
	at org.scalatest.tools.Runner$.doRunRunRunDaDoRunRun(Runner.scala:1343)
	at org.scalatest.tools.Runner$.$anonfun$runOptionallyWithPassFailReporter$24(Runner.scala:1033)
	at org.scalatest.tools.Runner$.$anonfun$runOptionallyWithPassFailReporter$24$adapted(Runner.scala:1011)
	at org.scalatest.tools.Runner$.withClassLoaderAndDispatchReporter(Runner.scala:1509)
	at org.scalatest.tools.Runner$.runOptionallyWithPassFailReporter(Runner.scala:1011)
	at org.scalatest.tools.Runner$.run(Runner.scala:850)
	at org.scalatest.tools.Runner.run(Runner.scala)
	at org.jetbrains.plugins.scala.testingSupport.scalaTest.ScalaTestRunner.runScalaTest2(ScalaTestRunner.java:133)
	at org.jetbrains.plugins.scala.testingSupport.scalaTest.ScalaTestRunner.main(ScalaTestRunner.java:27)

Process finished with exit code 0
```

HaFred, 2020-07-24:
Maybe there are some other dependency issues, if I did not bump the latest chisel-testers2, the exceptions are as below,
```
[info] [0.004] Elaborating design...
[info] [1.922] Done elaborating.
Computed transform order in: 273.8 ms
Total FIRRTL Compile Time: 2223.3 ms

An exception or error caused a run to abort: org.json4s.FullTypeHints.<init>(Lscala/collection/immutable/List;)V 
java.lang.NoSuchMethodError: org.json4s.FullTypeHints.<init>(Lscala/collection/immutable/List;)V
	at firrtl.annotations.JsonProtocol$.jsonFormat(JsonProtocol.scala:132)
	at firrtl.annotations.JsonProtocol$.serializeTry(JsonProtocol.scala:152)
	at firrtl.annotations.JsonProtocol$.serialize(JsonProtocol.scala:144)
	at firrtl.options.phases.WriteOutputAnnotations.transform(WriteOutputAnnotations.scala:38)
	at firrtl.options.phases.WriteOutputAnnotations.transform(WriteOutputAnnotations.scala:15)
	at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:37)
	at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:15)
	at firrtl.options.Translator.transform(Phase.scala:240)
	at firrtl.options.Translator.transform$(Phase.scala:240)
	at firrtl.options.phases.DeletedWrapper.transform(DeletedWrapper.scala:15)
	at firrtl.options.Stage.$anonfun$transform$5(Stage.scala:46)
	at scala.collection.LinearSeqOptimized.foldLeft(LinearSeqOptimized.scala:126)
	at scala.collection.LinearSeqOptimized.foldLeft$(LinearSeqOptimized.scala:122)
	at scala.collection.immutable.List.foldLeft(List.scala:89)
	at firrtl.options.Stage.$anonfun$transform$3(Stage.scala:46)
	at logger.Logger$.$anonfun$makeScope$2(Logger.scala:168)
	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:62)
	at logger.Logger$.makeScope(Logger.scala:166)
	at firrtl.options.Stage.transform(Stage.scala:46)
	at chisel3.stage.phases.MaybeFirrtlStage.$anonfun$transform$2(MaybeFirrtlStage.scala:19)
	at scala.Option.getOrElse(Option.scala:189)
	at chisel3.stage.phases.MaybeFirrtlStage.transform(MaybeFirrtlStage.scala:19)
	at chisel3.stage.phases.MaybeFirrtlStage.transform(MaybeFirrtlStage.scala:13)
	at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:37)
	at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:15)
	at firrtl.options.Translator.transform(Phase.scala:240)
	at firrtl.options.Translator.transform$(Phase.scala:240)
	at firrtl.options.phases.DeletedWrapper.transform(DeletedWrapper.scala:15)
	at firrtl.options.DependencyManager.$anonfun$transform$3(DependencyManager.scala:262)
	at scala.collection.LinearSeqOptimized.foldLeft(LinearSeqOptimized.scala:126)
	at scala.collection.LinearSeqOptimized.foldLeft$(LinearSeqOptimized.scala:122)
	at scala.collection.immutable.List.foldLeft(List.scala:89)
	at firrtl.options.DependencyManager.transform(DependencyManager.scala:255)
	at firrtl.options.DependencyManager.transform$(DependencyManager.scala:240)
	at firrtl.options.PhaseManager.transform(DependencyManager.scala:419)
	at chisel3.stage.ChiselStage.run(ChiselStage.scala:35)
	at chiseltest.legacy.backends.verilator.VerilatorExecutive$.start(VerilatorExecutive.scala:61)
	at chiseltest.defaults.package$.createDefaultTester(defaults.scala:24)
	at chiseltest.ChiselScalatestTester$TestBuilder.apply(ChiselScalatestTester.scala:30)
	at dla.tests.petest.ProcessingElementSpecTest.$anonfun$new$11(ProcessingElementSpecTest.scala:674)
	at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:23)
	at org.scalatest.OutcomeOf.outcomeOf(OutcomeOf.scala:85)
	at org.scalatest.OutcomeOf.outcomeOf$(OutcomeOf.scala:83)
	at org.scalatest.OutcomeOf$.outcomeOf(OutcomeOf.scala:104)
	at org.scalatest.Transformer.apply(Transformer.scala:22)
	at org.scalatest.Transformer.apply(Transformer.scala:20)
	at org.scalatest.FlatSpecLike$$anon$5.apply(FlatSpecLike.scala:1682)
	at org.scalatest.TestSuite.withFixture(TestSuite.scala:196)
	at org.scalatest.TestSuite.withFixture$(TestSuite.scala:195)
	at dla.tests.petest.ProcessingElementSpecTest.chiseltest$ChiselScalatestTester$$super$withFixture(ProcessingElementSpecTest.scala:488)
	at chiseltest.ChiselScalatestTester.$anonfun$withFixture$1(ChiselScalatestTester.scala:47)
	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:62)
	at chiseltest.ChiselScalatestTester.withFixture(ChiselScalatestTester.scala:47)
	at chiseltest.ChiselScalatestTester.withFixture$(ChiselScalatestTester.scala:44)
	at dla.tests.petest.ProcessingElementSpecTest.withFixture(ProcessingElementSpecTest.scala:488)
	at org.scalatest.FlatSpecLike.invokeWithFixture$1(FlatSpecLike.scala:1680)
	at org.scalatest.FlatSpecLike.$anonfun$runTest$1(FlatSpecLike.scala:1692)
	at org.scalatest.SuperEngine.runTestImpl(Engine.scala:286)
	at org.scalatest.FlatSpecLike.runTest(FlatSpecLike.scala:1692)
	at org.scalatest.FlatSpecLike.runTest$(FlatSpecLike.scala:1674)
	at org.scalatest.FlatSpec.runTest(FlatSpec.scala:1685)
[... 12 lines cut ...]
	at org.scalatest.FlatSpec.runTests(FlatSpec.scala:1685)
	at org.scalatest.Suite.run(Suite.scala:1124)
	at org.scalatest.Suite.run$(Suite.scala:1106)
	at org.scalatest.FlatSpec.org$scalatest$FlatSpecLike$$super$run(FlatSpec.scala:1685)
	at org.scalatest.FlatSpecLike.$anonfun$run$1(FlatSpecLike.scala:1795)
	at org.scalatest.SuperEngine.runImpl(Engine.scala:518)
	at org.scalatest.FlatSpecLike.run(FlatSpecLike.scala:1795)
	at org.scalatest.FlatSpecLike.run$(FlatSpecLike.scala:1793)
	at org.scalatest.FlatSpec.run(FlatSpec.scala:1685)
[... cut ...]
	at org.scalatest.tools.Runner$.withClassLoaderAndDispatchReporter(Runner.scala:1509)
	at org.scalatest.tools.Runner$.runOptionallyWithPassFailReporter(Runner.scala:1011)
	at org.scalatest.tools.Runner$.run(Runner.scala:850)
	at org.scalatest.tools.Runner.run(Runner.scala)
	at org.jetbrains.plugins.scala.testingSupport.scalaTest.ScalaTestRunner.runScalaTest2(ScalaTestRunner.java:133)
	at org.jetbrains.plugins.scala.testingSupport.scalaTest.ScalaTestRunner.main(ScalaTestRunner.java:27)

Process finished with exit code 0
```

sequencer, 2020-07-27:
I think this is not related to tester2?
if you are using https://github.com/sequencer/rocket-playground/ you can send issue there.

HaFred, 2020-07-28:
> I think this is not related to tester2?
> if you are using https://github.com/sequencer/rocket-playground/ you can send issue there.

Okay thank you.

closed 2020-07-28 by HaFred