ucb-bar/chiseltest #180: Is SInt available for expect?

HaFred, 2020-07-30:
When I tried to expect a SInt signal, the error showed up as below
```
Error:(25, 25) value expect is not a member of chisel3.SInt
        io.weightOut(i).expect(weightOutGoldenModel(i).U, s"the weightOut $i is not matched")
```

chick, 2020-07-30:
That should not be a problem, here is a slightly modified version of PrintfSpec from chiseltest's test suite and it works for me
```scala
class PrintfSpec extends FreeSpec with ChiselScalatestTester {
  "printf with negative hex numbers" in {
    test(new MultiIOModule() {
      val in = IO(Input(SInt(8.W)))
      val out = IO(Output(Vec(2, SInt(8.W))))

      printf("bitWidth=8 %x\n", in)
      out(0) := in
      out(1) := in + in
    }).withAnnotations(Seq(VerilatorBackendAnnotation)) { dut =>
      val (start, stop) = treadle.extremaOfSIntOfWidth(8)
      for(i <- start to stop) {
        dut.in.poke(i.S)
        dut.clock.step()
        println(s"dut.out ${dut.out(0).peek()} ${dut.out(1).peek()}")
        dut.out(0).expect(i.S)
      }
    }
  }
}
```
Can you share more of the code, at least the declaration of `io.weight`

ducky64, 2020-08-02:
SInt operations should be supported, as in [ElementTest.scala:53](https://github.com/ucb-bar/chisel-testers2/blob/master/src/test/scala/chiseltest/tests/ElementTest.scala#L53). You may have to make sure that the argument to expect is the same type as the signal being expected on (so if `io.weightOut(i)` is SInt, you would use `io.weightOut(i).expect(weightOutGoldenModel(i).S, ...)` (note the `.S` instead of `.U` in the `.expect`). Also, did you make sure to `import chiseltest._`?

closed 2020-08-02 by HaFred