ucb-bar/chiseltest #183: Minor additions: step and joinAll

schoeberl, 2020-08-06:
Two small additions would be nice:

```
dut.clock.step()
```
is a bit long to type, just a ```step()``` would be nice and short.
A function to join all parallel threads, so the following pattern would work:
```
fork {
  testA
}
fork {
  testB
}

joinAll()

edwardcwang, 2020-08-06:
I think one possible motivation for avoiding `step()`, is that in a design with multiple clocks, it creates an ambiguity as to which clock to step.

schoeberl, 2020-08-06:
But if there is only one clock (99% of designs/ unit tests) it would work.

schoeberl, 2020-08-06:
Related to the clock request: access to a clock tick counter, e.g.,  #```clock.count()```, so I do not need to do this manually in each of my threads.

Cheers,
Martin

ducky64, 2020-08-06:
I'm generally not a fan of doing operations where the scope is clear, so I don't think a global `step()` is a good idea. But, if we were really concerned about saving a few characters, it might make sense to allow a `step` on an ImplicitModule, since the clock there is less ambiguous. So instead of `dut.clock.step()`, you could just have `dut.step()`.

One possible downside is that even ImplicitModules can have multiple clocks, though the behavior is much less "global" than a bare `step()`.

Thoughts?

Clock count is discussed in #71. As a slight recap, I'm not sure count() with ticks-since-beginning makes sense, since we do a few clock steps with reset asserted before entering user testdriver code, and I'm not a huge fan of global state. It might make sense to have an API like clock.counter() that returns an object that can be queried for the ticks since instantiated.

ducky64, 2020-08-06:
Also, how do you see joinAll working? I could see a few options:
- join on all threads in existence, which might be undesirable if there were threads that aren't supposed to end
- join on all threads in the current lexical scope, which would probably require macros and compiler plugins
- join on all threads that are a child (or more generally, a descendant?) of the current thread

I think unless there's broad agreement on semantics and example use cases which demonstrates the power beyond saving a few characters, the downsides of potential confusion and non-local behavior outweigh the benefits?

schoeberl, 2020-08-07:
Ok, I proposed this because I was stumbling over the issue in my testing code. So there is a use case. I guess when you need a parallel testing thread chances are high you need more than two. How would you join more than one? Example code:
```
for (i <- 0 until nrOfPorts) {
  fork {
      doSomeWrites(i)
  }
  fork {
    receiveTheWrites(i)
  }
}
// Want to wait until all tests finished and print results
// How do I join all those?
```

Join all threads or all that are childs would be a solution. Another option is to get a thread reference back on fork and using that to join (like in the Java Thread).

Cheers,
Martin

schoeberl, 2020-08-07:
What is ```fork``` returning? From the tester code I see a chiseltest.internal.TesterThreadList. Can I use this? Wouldn't like to use something that is internal.

ducky64, 2020-08-07:
Yeah, `fork` returns the spawned thread (well, the ChiselTest wrapper around the thread object). If you call a `fork` off that, eg `fork { ... }.fork { ... } `, the returned object from the second `fork` includes both threads. You can see an example in [ThreadJoinTest.scala](https://github.com/ucb-bar/chisel-testers2/blob/master/src/test/scala/chiseltest/tests/ThreadJoinTest.scala#L40)

ducky64, 2020-08-14:
Does the existing fork/join and thread-list API work for you, or is there something still missing?

Clock counters is a separate issue, but what are people's thoughts on Module.step() as a shorthand for Module.clock.step()? Typically written, it would be more like `c.step()` instead of `c.clock.step()`?