ucb-bar/chiseltest #184: Shared state between threads

schoeberl, 2020-08-06:
Is there a way to safely share state between thread, synchronized with step().

I know that each step() synchronizes all tester threads, but the individual threads continue in whatever order until the next step(). However, if I need to update a shared variable and I need to coordinate this with step(), I need some function that may run immediately after all threads synchronized on step(). Maybe this sounds a bit confusing. Here an example:
```
var cnt = 0

for (i <- 0 until nrOfThreads) {
fork {
  for (...) {
      println(cnt)
  }
}
}
dut.clock.stepAndExecuteAfterAllDidTheStep(1) {
  cnt += 1
}

Or is there any Scala barrier function that I could use? 

Cheers,
Martin

ducky64, 2020-08-06:
So the test thread behavior is actually cooperative multitasked - they are run one after the other by a scheduler, changing threads only during `step()` (which acts as a scheduler call / blocking wait). There is a deterministic order to thread execution, and ordering can be further controlled using [regions](https://github.com/ucb-bar/chisel-testers2/blob/master/src/test/scala/chiseltest/tests/RegionsTest.scala) (threads can have an associated region; within each time step, all threads in one region are run before the next; there are currently two regions: default and `Monitor`, with default running before `Monitor`).

Nondeterministic race conditions in the usual sense should not be possible with the current implementation, but if you don't factor in thread execution order, you may get deterministic but nonintuitive behavior. I'd recommend the use of regions since it's an API designed to make sure certain threads run before / after others, but you can also rely on the thread execution order.

schoeberl, 2020-08-07:
Thanks, this looks like what I need.

closed 2020-08-07 by schoeberl