ucb-bar/chiseltest #186: (VCD) options

schoeberl, 2020-08-07:
In one example I see ```test(new Module(...)).withAnnotations(Seq(WriteVcdAnnotation)) { ... }``` to pass an option to generate a VCD. However, in user land I cannot access ```.withAnnotations```. Am I missing an import? Or is there a different way to pass annotations?

Cheers,
Martin

ducky64, 2020-08-07:
As in the [VerilatorBasicTests.scala](https://github.com/ucb-bar/chisel-testers2/blob/master/src/test/scala/chiseltest/experimental/tests/VerilatorBasicTests.scala) example. I think you need `import chiseltest.experimental.TestOptionBuilder._` to use `.withAnnotations`

chick, 2020-08-09:
And make sure to use the `chiseltest.internal.WriteVcdAnnotation`, there is also a `treadle.WriteVcdAnnotation` that will only work when using the treadle backend. (historical awkwardness)

schoeberl, 2020-08-13:
Thanks, both of you for the info. Works on my side. I guess the API should change to avoid ```internal``` and ```experimental``` package names. Waveforms are a fundamental debugging tool for hardware designers.

If this is not covered by another issue, I leave this issue open as a reminder.

schoeberl, 2021-08-11:
I am updating the testing section for the third edition of the Chisel book to ChiselTest. And having the following code in a textbook does not look like we are using a mature tool:
```
import chiseltest.experimental.TestOptionBuilder._
import chiseltest.internal.WriteVcdAnnotation
```
Is there any chance to move (or copy) this to the base ```chiseltest``` package? Could this be done for Chisel 3.5, which I am targeting with the 3rd edition?

schoeberl, 2021-08-11:
Ok, this would be great!

I am not telling my students anything about flags or annotations if not needed. They are second semester students with no programming background at all! I need to keep it as simple as possible. But the last two years it worked out ok.

ekiwi, 2021-08-11:
> I am not telling my students anything about flags or annotations if not needed. They are second semester students with no programming background at all! I need to keep it as simple as possible. But the last two years it worked out ok.

Great! Just wanted to make sure that it is enough to make the `withAnnotations` public and we don't have to care about the `withFlags`.

ekiwi, 2021-12-18:
fixed

closed 2021-12-18 by ekiwi