ucb-bar/chiseltest #189: How to cite tester2

schoeberl, 2020-08-13:
I am about to submit a paper where I am using tester2/ChiselTest in the evaluation and would like to correctly cite this work.

Cheers,
Martin

ducky64, 2020-08-14:
Uh, since there's no publication on this (yet?), just drop the GitHub link?

And at some point we'll probably get around to changing all the naming to ChiselTest, but I think GitHub is pretty good about repository redirects?

schoeberl, 2020-08-14:
What about having a technical report published by UCB? A GitHub link is a very weak reference.

Yes, getting the renaming done would be good. And than call it version 1.0 and be it the official Chisel testing framework ;-)

Cheers,
Martin

ducky64, 2020-08-14:
A technical report might be a possibility, but probably not a near-term one.

I'm actually not sure what is holding up the repo rename. It looks like we've already renamed the package in build.sbt (though I think the publish scripts continue to also publish under chisel-testers2 for compatibility?).

schoeberl, 2020-09-22:
I am using the following and will close this.
```
@ELECTRONIC{chisel:tester2,
  author = {Richard Lin},
  title = {{ChiselTest}},
  howpublished = {https://github.com/ucb-bar/chisel-testers2}
}
```

closed 2020-09-22 by schoeberl