ucb-bar/chiseltest #192: Generate verilog with chiseltest

hadirkhan10, 2020-08-24:
Hello, I previously used chisel3.iotesters for writing testbenches and invoked with the following command to generate `.vcd` as well as `.v` verilog files. How is this possible with chiseltest. I can generate `.vcd` files using:
```
testOnly testclassName -- -DwriteVcd=1
```
But how can one generate a verilog file in the `test_run_dir` as well?

ducky64, 2020-08-25:
You can write tests so they run under Verilator. See the [usage references in the README](https://github.com/ucb-bar/chisel-testers2#usage-references). Otherwise, tests by default run under treadle, which operates on the FIRRTL IR and doesn't need (so we don't generate) the Verilog files. There currently isn't a way to change the default backend (aside from modifying [defaults.scala](https://github.com/ucb-bar/chisel-testers2/blob/master/src/main/scala/chiseltest/defaults.scala)), but maybe that should be a command line flag?

hadirkhan10, 2020-08-26:
Yes I was able to generate verilog with the Chisel Stage

closed 2020-08-26 by hadirkhan10