ucb-bar/chiseltest #202: Cuncrrency question

schoeberl, 2020-09-22:
This is more a clarification question instead of an issue. Is it correct that fork actually implements somthing like a coroutine, meaning that all forked code is executed sequentically and the switch between the code blocks happens on step()? Threads are only used to implement this functionallity and nothing is reall running in parllel.

Therefore, is it safe to do following:
```
var cnt = 0

fork {
  for (...) {
    cnt  += 1
    c.step()
  }
}

for (...) {
  cnt  += 1
  c.step()
}
```

Meaning there is no race condition on the increment of ```cnt```.

edwardcwang, 2020-09-22:
Do you mean something like this?

```
fork {
  for (...) {
    cnt  += 1
    c.step()
  }
} .fork {
  for (...) {
    cnt  += 1
    c.step()
  }
} .join
```

schoeberl, 2020-09-22:
This is similar to my example, but you create 2 additional threads. Mine has just two threads. But the question is related to the concurrent update of cnt, not the number of threads.

edwardcwang, 2020-09-22:
Well in the original example there only appears to be a single thread, since there is only 1 `fork`? The second for loop appears to just be executed sequentially after the first loop.

schoeberl, 2020-09-22:
The fork creates a second thread. So there is the main thread and the forked thread. Anyway, this is not about my question (I did in my use case a for loop with forks, so there are many threads). The question is (probably to Richard) if this concurrency model is like coroutines without real parallelism and we do not need to protect our data with locks.

ducky64, 2020-09-23:
@edwardcwang fork doesn't block until you `.join()` on the created thread, so while I think your example is stylistically more elegant, both are "threaded".

@schoeberl Yes, the concurrency model is like coroutines, and there should be no real parallelism inbetween constructs that advance time (like `step`, which invokes the scheduler). The intent of this programming model is to allow concurrency in testing naturally parallel hardware and data operations, but without the confusion and race conditions potentially arising from actual parallelism. There is a defined thread order, but data flows through the simulator (peek-after-poke, etc) are checked for certain cross-thread combinational paths. Data flows outside the simulator can't be checked (at least without fancy things like compiler extensions), so it's kind of a free-for-all.

There is a defined thread execution order, which (if I remember correctly) largely follows the program text (spawned threads run immediately, then return to their parent; on subsequent cycles child threads run before their parent, in the order they are spawned). But you should still be disciplined about (and avoid if possible) cross-thread operations, since we can't check those, and you have to reason about thread order (if that affects correctness for your use case). Note that traditional synchronization constructs probably won't be too useful, since there's no actual parallelism.

schoeberl, 2020-09-23:
Thank you for the detailed explanation! Maybe this should be added to the documentation (README). @ducky64 should decide on closing this issue. I am done with it.

PR #211 (2020-10-10): Add thread ordering documentation and supporting test
As described in the title, adds the fork behavior to the README, and a test for it.

Resolves #202
closed 2020-10-10 by ducky64