ucb-bar/chiseltest #214: Parallel Test Enhancement

SihaoLiu, 2020-10-15:
The current Chisel Testers2 will only do sequential test when I write:
```
val hw_para_seq : Seq[Int] = Seq(1, 2, 3, 4, 5, 6)
for(hw_para <- hw_para_seq){
    test(new Module(hw_para)){dut=>
       // do test here
    }
}
```
The purpose of this is that I want to test the hardware Module I built under different parameters (like input data width = 8/16/32/64). I am pretty sure that there is no dependency between the loop. But for some reason, this cannot be parallelized.

I also tried parallel sequence and used `map` or `foreach` but a dependency error was thrown. The only way I can do parallel test is I wrote in this way:
```
"test 1 in parameter 1" in {
     test(){dut => }
}
"test 2 in parameter 2" in {
     test(){dut => }
}
"test 3 in parameter 3" in {
     test(){dut => }
}
```
But this asks me to write a lot of repeative code
Is it possible to add parallel test support in Chisel Testers2?

ducky64, 2020-10-15:
So ChiselTest only handles simulator construction (including elaboration and compilation) and interface (by providing APIs for peek and poke). Defining and running test cases is out of scope (and is handled by other libraries like ScalaTest, which ChiselTest provides integration points for).

With regards to writing DRY code, you can consider abstracting `test(){dut => }` into a function, so your tests look like
```scala
"test 1 in parameter 1" in {
    myTestFn(1)
}
"test 2 in parameter 2" in {
    myTestFn(2)
}
```
Also, consider looking at #196, which uses a ScalaTest construct to programmatically construct test cases. I don't know if it does true parallelism, though.

Also, do you have example code that produces a dependency error? (I don't think it's an error, but would like to understand how you understood the APIs)

SihaoLiu, 2020-10-15:
The problem here is that the amount of hardware parameters is too many since I want to test all combination of all possible hardware parameters like below:
```
val hardware_parameter1_seq = Seq(1, 2, 3, 4, 5, 6, 7, 8)
val hardware_parameter2_seq = Seq(2, 4, 6, 8, 10, 12, 14)

val hardware_parameter_combination = 
    for(hw_para1 <- hardware_parameter1_seq;
         hw_para2 <- hardware_parameter2_seq) yield {
        (hw_para1, hw_para2)
}

// it would be nice to have
hardware_parameter_combination.foreach {
    (hw_para1, hw_para2) => 
        test(new Module(hw_para1, hw_para2)){
            // do test here
        }
}
```
Instead of write 7 x 8 = 56 lines of code:
```
"test1: hw para1 = 1, hw para2 = 2" in {  myTestFn(...)  }
"test2: hw para1 = 1, hw para2 = 4" in {  myTestFn(...)  }
...
"test56: hw para1 = 8, hw para2 = 14" in {  myTestFn(...)  }
```

You have any suggestion on how to do this in parallel instead of writing that many lines of code?

ducky64, 2020-10-15:
I see what you're trying to do now. Have you looked into whether ScalaTest supports parallel test execution, such as this [StackOverflow post](https://stackoverflow.com/questions/15752681/run-scalatest-tests-in-parallel)? (quick Googling also returned some old documentation for ParallelTestExecution; overall this doesn't seem to be a common use case)

Otherwise, you can probably use Scala parallelism to put all the parallel `test(...) {...}` invocations into one big ScalaTest test. But I don't think it's necessary (or makes sense, from a separation of concerns standpoint) to have parallelization as a construct in ChiselTest rather than in the unit test framework like ScalaTest.

ekiwi, 2020-11-13:
You can always programmatically generate test cases.
I use this to test an ALU that I wrote: https://github.com/ekiwi/serv-chisel/blob/8f11fa298371a4257de4e1d169a39e0b97dfcc4f/test/serv/AluSpec.scala#L59

closed 2021-04-07 by ekiwi