ucb-bar/chiseltest #217: Basic Question: how to run something other than treadle

mwachs5, 2020-10-29:
I'll make a PR to update the README if someone can explain / point me to this answer (a pointer to the relevant code is fine)...

the README says "`test` automatically runs the default simulator (which is treadle), and runs the test stimulus in the block"

How do I change that behavior (to use VCS, verilator)?

sequencer, 2020-10-29:
There are annotations:
`VcsBackendAnnotation` and `VerilatorBackendAnnotation`, but I think we should document it.

chick, 2020-10-29:
@mwachs5 
Take a quick look at how the verilator tests work in chiseltest's [VerilatorBasicTests](https://github.com/ucb-bar/chisel-testers2/blob/master/src/test/scala/chiseltest/experimental/tests/VerilatorBasicTests.scala)
You must use the `import chiseltest.experimental.TestOptionBuilder._`
to get `.withAnnotation`, there is also a `.withFlags` to use command line options style

mwachs5, 2020-10-29:
thanks for the pointers! I will take a look

mwachs5, 2020-10-29:
Thanks. Notes for myself:
The annotation sets the "BackendExecutive" which has the guts of launching the backend:
 https://github.com/ucb-bar/chisel-testers2/blob/3e3ecc5b25b7b6bc48341ec07c7a54b7ad53bcb7/src/main/scala/chiseltest/legacy/backends/verilator/VerilatorExecutive.scala#L15

chick, 2020-12-30:
I think this can be closed now

closed 2020-12-30 by chick