ucb-bar/chiseltest #219: Simulation mismatch vs. Icarus verilog

hutch31, 2020-11-17:
I have written a synthesizable testbench in chisel and exercise it with chiseltest.  While debugging my design, I noticed incorrect simulation behavior (a Queue object deasserts enq.ready when enq.valid was not asserted).  I dumped the testbench out to Verilog and simulated in Icarus, where I get the expected behavior.

The attached images show the resulting waveform from chiseltest (top) and icarus (bottom).

The source code and testbench are available here: https://github.com/hutch31/chisel-packetlib/tree/bugreport-201117

[image]

Zip containing VCD files from both simulation runs:
[env.zip](https://github.com/ucb-bar/chisel-testers2/files/5556871/env.zip)

ducky64, 2020-11-18:
If the stimulus being sent into the circuit is equivalent in both cases, this might be a treadle (simulator) issue instead of a chiseltest (stimulus generation / description language) issue? @chick (who develops treadle) - thoughts?

That being said, if the testbench stimulus is different, the it's possibly a chiseltest issue.

Note that there may be slightly different assumptions between Chisel and Verilog code, particularly around X handling. Because Chisel and treadle don't have a concept of a logic-invalid value (eg, X), treadle may default initialize or randomize undriven signals. If the result is nondeterministic, this may be related to randomization initialization.

Anyways, given the above screenshots, it's not immediately obvious where the problem is. The waveforms are pretty big, so it's difficult to trace through, especially without understanding the underlying circuit. A minimal example might be helpful if further debugging is needed.

hutch31, 2020-11-18:
The stimulus to the two circuits is nearly identical, as the external Verilog testbench simply provides reset and a clock and then triggers the Chisel testbench to start traffic.

I can try to cut this testcase down, but given that the misbehaving component is Queue, which is quite simple, I suspect that context is important here.

chick, 2020-11-18:
I will take a look at as soon as I can. It might be a treadle issue

chick, 2020-12-30:
@hutch31 
Sorry to take so long to get back to this. I cloned the chisel-packetlib repo and am close to getting things to run but I cannot find `CreditIO`, `DCCrossbar`, `DCCreditSender`, `DCCreditReceiver` are these in another repo?

hutch31, 2020-12-31:
@chick 
Hi Chick, yes those items are in https://github.com/hutch31/ip-contributions.  I believe Martin just merged my latest pull request with the master, so the mainline (https://github.com/freechipsproject/ip-contributions) should be up-to-date.

chisel-packetlib should have ip-contributions as a submodule.

chick, 2021-01-01:
@hutch31
I got everything running and this is the VCD trace I get from the treadle backend run
[image]
and this one is from the verilator backend.
[image]
I think these both look correct. This is running with current masters, which for treadle should be the same as the 3.4/1.3 release of chisel3/treadle.
Does this look correct to you

hutch31, 2021-01-02:
Yes these both look correct, the original error waveform had ready being
deasserted without seeing a valid.

closed 2021-03-20 by ekiwi