ucb-bar/chiseltest #220: Timescope signal

Vict0rAH, 2020-11-19:
Hi, I have a question concerning the use of timescope in the code shown below. What I am trying to do is making a concurrent assertion and poke a new value that would make the assertion fail, see the following:
```scala
dut.io.s.poke(4.U)
dut.clock.step(1)
fork {
    for (i <- 0 until 20) {
        assert(dut.io.c.peek.litValue() == 4, "Error")
        dut.clock.step(1)
    }
}
//timescope {
    dut.clock.step(10)
    dut.io.s.poke(0.U)
//}
```
When I run the above in my test, I receive an ThreadOrderDependentException stating that "Timescope signal changed, poked after peek". Wrapping the two lines of code after the fork inside a timescope (commented out in the above) resolves the issue, but should it not be possible to write this without the use of timescope?

Kind regards,
Victor

ducky64, 2020-11-19:
If there is a combinational dependency from `s` -> `c`, then chiseltest is designed to prevent you from doing conflicting writes, or read-after-write operations from different threads on the same clock cycle, because it now depends on the order threads run in which can be unintuitive (though it is deterministic). There are a few exceptions, including that you can explicitly communicate ordering using regions, which can help implement the monitor pattern.

If you have a different use case for this, I'm curious to hear about it.

I don't know why adding a timescope removes the error, that's probably a bug.

Vict0rAH, 2020-11-21:
Hi again, makes sense, after making a register inside the main class, the code works fine without a timescope. Thanks for your help.

closed 2020-11-21 by Vict0rAH