ucb-bar/chiseltest #237: Question about ending and old timescopes

hansemandse, 2020-11-25:
Hi. I am currently working on a small AXI4 testing framework with a BFM for the AXI4 master supposed to abstract things to a transaction level. Currently, each channel is handled by a separate thread that is spawned whenever there is work to be done on the channel, and each of the threads spawn new threads for managing other channels required to finish the transactions (i.e., a write address handler spawns a write handler etc.). 

To enfore correct execution order, I am trying to force spawned threads to wait for the most recently spawned thread of the same kind like shown below. However, I am getting a `chiseltest.ThreadOrderDependentException: Mix of ending timescopes and old timescopes` error. Can you point me in the right direction to solve this?

```scala
/** Fork new write handler */
lastWT = fork {
  lastWT.join()
  writeHandler()
}
```

My entire source is available [here](https://github.com/chisel-uvm/class2020/blob/master/axi4/src/main/scala/axi4/AXI4FunctionalMaster.scala).

Thanks in advance!

ducky64, 2020-11-27:
So one rule of threading here is that threads must be encapsulating - so a parent thread must not end before its children ends (this is more of an artifact for timescopes to be encapsulating - so there's a well-defined value to revert to - and fork creates a new timescope so that any signals poked in the 'top-level' of the fork are reverted after the thread ends).

For your snippet, can you write this instead?
```scala
lastWT.join()
lastWT = fork {
  writeHandler()
}
```

If you need the main thread to not block on the join, this probably won't work. In that case, could you have all the synchronized stuff run on its own thread? Otherwise, we'll have to think about your use case - inter-thread communication between sibling threads (without the parent orchestrating) isn't great.

hansemandse, 2020-11-28:
I agree with your points! Therefore, I have redesigned my BFM to not spawn threads from within child threads. It seems to work, so no need to consider my previous approach.

Thanks for helping!

closed 2020-11-28 by hansemandse