ucb-bar/chiseltest #241: VCS flags var is not being used in VcsExecutive

chick, 2020-12-01:
`moreVcsFlags` defined at line 87 in VcsExecutive 
```scala
    val moreVcsFlags = compiledAnnotations.collectFirst { case VcsFlags(flagSeq) => flagSeq }
```
is not being used.
I believe it should be incorporated at line 106 like this
```scala
    val vcsFlags = moreVcsFlags + moreVcsCFlags ++ coverageFlags
```
which would be similar to the usage in VerilatorExecutive.

chick, 2020-12-01:
@TsaiAnson can you tell by looking if this is broken as I believe it is.

TsaiAnson, 2020-12-01:
@chick Yeah that doesn't seem to be used. I agree that it should be added to line 106 but with a `++` op:
```scala
val vcsFlags = moreVcsFlags ++ moreVcsCFlags ++ coverageFlags
```

chick, 2020-12-30:
I believe this is fixed in PR #242

closed 2020-12-30 by chick