ucb-bar/chiseltest #268: No new subfolder for every new test run

schoeberl, 2021-02-25:
When working with waveforms, which is quite common for hardware debugging, it is very confusing that each test run generates a new folder. I foresee my students struggling in the next weeks staring at the wrong (outdated) waveforms due to this. And you need to reopen the waveform each time. When overwriting the old version, one can simply reload in GTKWave.

edwardcwang, 2021-02-25:
Is this with testers1 or testers2? With testers2 the `test_run_dir` doesn't seem to have any timestamped entries for me.

ducky64, 2021-02-27:
I just tested it on latest and:
- With ScalaTest-based tests, the folders in test_run_dir are stable.
- However, RawTester-based tests (only `chiseltest.tests.NoScalatestTesterTest`, I think) do not have access to the test name (there might not even be one) or circuit name, so it just creates a folder called `s"chisel_test_${System.currentTimeMillis()}"` as in [RawTester.scala:56](https://github.com/ucb-bar/chisel-testers2/blob/74094850f712a68949c55bae1e92f7d57051710e/src/main/scala/chiseltest/RawTester.scala#L56). Not sure if there's a good way to solve this, though.

chick, 2021-02-27:
@schoeberl RawTester#test takes an `Seq[Annotation]` as a second argument. So the following will create a stable
test directory
```
test(new ShifterModule(UInt(8.W), 4), Seq(TargetDirAnnotation("test_run_dir/raw_test"))) { c =>
```
Given the hoops we have to jump through to get the dir named by the dut name, I am moving towards the conclusion that all tests should be explicitly named by their creators

schoeberl, 2021-03-03:
Oh sorry, my fault. It was an iotesters test, as I have not yet switched to ChiselTest with my teaching material.
Please ignore it. @chick Thanks for the pointer and I agree (with waveforms) that one probably wants to have explicit folder names.

closed 2021-03-03 by schoeberl