ucb-bar/chiseltest #273: Passing CLI Options to Verilator

BracketMaster, 2021-03-14:
Is there an annotation that will allow me to pass a command line
option to the verilator invocation?

Perusing through the source, this isn't immediately obvious.
I currently do annotations [this way](https://github.com/BracketMaster/chisel-by-example/blob/78bb5cdc51d5032e08c6414049196bbc8bfb3dcc/Counter/test/src/CounterTest.scala#L16),
which I believe is the prescribed way to do it

jackkoenig, 2021-03-15:
As we discussed in dev today, We think `VerilatorFlags` is the annotation to use: https://github.com/ucb-bar/chisel-testers2/blob/2e676d3d1489787aa8700ad0d129f36ca0b74390/src/main/scala/chiseltest/legacy/backends/verilator/VerilatorAnnotations.scala#L29

BracketMaster, 2021-03-15:
Thanks. I got verilator multithreaded working with:

```scala
val threads = 2
import chiseltest.legacy.backends.verilator.VerilatorFlags
VerilatorFlags(Seq("--threads", s"${threads}")))
```

[Here's a real example](https://github.com/BracketMaster/chisel-by-example#verilator-multithreaded)
obviously multithreading a counter doesn't buy
you much, but for large simulations, the performance benefit is easily
justified.

closed 2021-03-15 by BracketMaster