ucb-bar/chiseltest #282: Verilator tests failing

hansemandse, 2021-03-30:
Some Verilator tests of mine are failing after updating to 0.5-SNAPSHOT. The tests in question are `neuroproc.unittests.TrueDualPortMemoryTester` and `neuroproc.unittests.TrueDualPortFIFOTester` available [here](https://github.com/hansemandse/KWSonSNN/tree/master/src/test/scala/neuroproc/unittests). Both tests are working in 0.3.2.

The error message is:
```[info]   java.lang.RuntimeException: Failed to find line `VL_COVER_INSERT(count32p,  "filename",filenamep,  "lineno",lineno,  "column",column,` in test_run_dir/True_dual_port_memory_should_work_in_both_Chisel_and_Verilog/VTrueDualPortMemoryVerilog.cpp.```

I am using:
- chisel3 3.5-SNAPSHOT ([7b8fbbe](https://github.com/chipsalliance/chisel3/commit/7b8fbbe8913a2eb92f50948dc2be50dc482fb58d))
- chisel-testers2 0.5-SNAPSHOT (5d5071a)

`verilator --version` => Verilator 3.916 2017-11-25 rev verilator_3_914-65-g0478dbd

ekiwi, 2021-03-30:
Thanks for the report.

To support the new cover points feature that will be introduced in the x.5 release, we will release we will be requiring at least verilator `4.026`.
The immediate cover statements that the firrtl compiler may produce now are only supported starting with [Verilator 4.026](https://www.veripool.org/news/266-Verilator-Verilator-4-026-Released).

However, I will try to see if there is a way to get older verilator versions supported as long as you are not using any cover points.

ekiwi, 2021-03-30:
Kevin's TODO:
- [ ] do not try to patch coverage when there are no cover points in the design
- [ ] detect verilator version and show a warning if it is too old

ekiwi, 2021-03-30:
> I'm seeing this issue with the latest Verilator from stable branch (Verilator 4.200 2021-03-12 rev v4.200-16-gf0d66453) as well.

Thanks for pointing that out. Do you happen to have a small test case that we could add to CI?
Our CI is currently using `Verilator 4.028 2020-02-06 rev v4.026-92-g890cecc1` and unfortunately did not catch this regression.

kammoh, 2021-03-30:
Sure, but I'm guessing the issue is not really related to Verilator version. 
Shouldn't coverage pass and data analysis only be evoked if some coverage annotations are actually included in the elaborated design?
I could be totally misunderstanding this, but the following seems to be working for me: https://github.com/ucb-bar/chisel-testers2/pull/283

ekiwi, 2021-03-30:
@hansemandse I tried to reproduce the error that you are seeing.

Unfortunately, `neuroproc.unittests.TrueDualPortMemoryTester` seems to work on my system and for `neuroproc.unittests.TrueDualPortFIFOTester` I get an error from Verilator which does not seem to like your Verilog:

```
%Error-PROCASSWIRE: /home/kevin/d/KWSonSNN/test_run_dir/True_dual_port_FIFO_should_flag_empty_and_full/TrueDualPortFIFOBB.v:19:3: Procedural assignment to wire, perhaps intended var (IEEE 1800-2017 6.5): 'wAddrNext'
                                                                                                                                : ... In instance TrueDualPortFIFOVerilog.fifo
   19 |   wAddrNext = wAddr == 7 ? 3'b000 : wAddr + 1;
      |   ^~~~~~~~~
%Error-PROCASSWIRE: /home/kevin/d/KWSonSNN/test_run_dir/True_dual_port_FIFO_should_flag_empty_and_full/TrueDualPortFIFOBB.v:20:3: Procedural assignment to wire, perhaps intended var (IEEE 1800-2017 6.5): 'rAddrNext'
                                                                                                                                : ... In instance TrueDualPortFIFOVerilog.fifo
   20 |   rAddrNext = rAddr == 7 ? 3'b000 : rAddr + 1;
      |   ^~~~~~~~~
%Error-PROCASSWIRE: /home/kevin/d/KWSonSNN/test_run_dir/True_dual_port_FIFO_should_flag_empty_and_full/TrueDualPortFIFOBB.v:22:3: Procedural assignment to wire, perhaps intended var (IEEE 1800-2017 6.5): 'full'
                                                                                                                                : ... In instance TrueDualPortFIFOVerilog.fifo
   22 |   full  = fullReg;
      |   ^~~~
%Error-PROCASSWIRE: /home/kevin/d/KWSonSNN/test_run_dir/True_dual_port_FIFO_should_flag_empty_and_full/TrueDualPortFIFOBB.v:23:3: Procedural assignment to wire, perhaps intended var (IEEE 1800-2017 6.5): 'empty'
                                                                                                                                : ... In instance TrueDualPortFIFOVerilog.fifo
   23 |   empty = emptyReg;
      |   ^~~~~
%Error-PROCASSWIRE: /home/kevin/d/KWSonSNN/test_run_dir/True_dual_port_FIFO_should_flag_empty_and_full/TrueDualPortFIFOBB.v:57:7: Procedural assignment to wire, perhaps intended var (IEEE 1800-2017 6.5): 'datao'
                                                                                                                                : ... In instance TrueDualPortFIFOVerilog.fifo
   57 |       datao <= ram[rAddr];
      |       ^~~~~
%Error: Exiting due to 5 error(s)
```

hansemandse, 2021-04-02:
@ekiwi Sorry for the delay and thanks for helping out.

I'm not yet very experienced in writing Verilog, so it is definitely possible that it is incorrect. Strange that Verilator 3 accepts it, though. I'll look at updating to Verilator 4.028 and fixing my design.

ekiwi, 2021-04-02:
> I'm not yet very experienced in writing Verilog, so it is definitely possible that it is incorrect. Strange that Verilator 3 accepts it, though. I'll look at updating to Verilator 4.028 and fixing my design.

Verilog tools traditionally accept a lot of non-standard-compliant Verilog code and every vendor has their custom dialect that they support. The verilator developers seems to have more faith in standards and are continuously adding more checks to see if the Verilog is standard compliant. It is very possible that the check that failed on the newer verilator version was added recently.

ekiwi, 2021-04-02:
> I'm seeing this issue with the latest Verilator from stable branch (Verilator 4.200 2021-03-12 rev v4.200-16-gf0d66453) as well.

@kammoh do you have any open source code that you can point me to that shows this issue? Since I could not reproduce it with @hansemandse's code, I am not sure what is going on.

kammoh, 2021-04-06:
@ekiwi Sorry for the late response.
Verilator tests from master branch fail on my machine:
```
[info] *** 20 TESTS FAILED ***
[info] utest
[info] Tests: 1, Passed: 1, Failed: 0
[error] Failed: Total 136, Failed 20, Errors 0, Passed 116, Ignored 4, Canceled 8
[error] Failed tests:
[error] 	chiseltest.experimental.tests.VerilatorClockPokeTest
[error] 	chiseltest.experimental.tests.VerilatorBasicTests
[error] 	chiseltest.experimental.tests.VerilatorCoverageTests
[error] 	chiseltest.coverage.VerilatorCoverageTest
[error] 	chiseltest.experimental.tests.NegativeInputValuesTest
[error] (Test / test) sbt.TestsFailedException: Tests unsuccessful
[error] Total time: 97 s (01:37), completed Apr 6, 2021, 2:54:53 PM
```

```
[info] VerilatorBasicTests:
[info] Testers2 with Verilator
[info] - should test static circuits *** FAILED ***
[info]   java.lang.RuntimeException: Failed to find line `VL_COVER_INSERT(count32p,  "filename",filenamep,  "lineno",lineno,  "column",column,` in test_run_dir/Testers2_with_Verilator_should_test_static_circuits/VStaticModule.cpp.
[info] Please file an issue and include the output of `verilator --version`
```

```
$ verilator --version
Verilator 4.200 2021-03-12 rev v4.200-16-gf0d66453
```

ekiwi, 2021-04-06:
Thanks @kammoh, you are definitely on to something. I just upgraded my local verilator to `4.108` and I can reproduce the problem. I am adding tests for a variety of different Verilator versions and it seems like currently only the narrow interval of `[4.028, 4.032, 4.034]` works: https://github.com/ucb-bar/chisel-testers2/pull/289

I will try to get the CI improvements merged and then I will enable the newer Verilator versions once I can fix the issue.

closed 2021-04-07 by ekiwi

hansemandse, 2021-04-08:
@ekiwi Sorry to bother you again, but after the addition of #290, one of the [CI tests](https://github.com/hansemandse/KWSonSNN/runs/2298553135) you helped me set up fails.

Runs with `verilator --version` => Verilator 4.028 2020-02-06 rev v4.026-92-g890cecc1

ekiwi, 2021-04-08:
> @ekiwi Sorry to bother you again, but after the addition of #290, one of the [CI tests](https://github.com/hansemandse/KWSonSNN/runs/2298553135) you helped me set up fails.
> 
> Runs with `verilator --version` => Verilator 4.028 2020-02-06 rev v4.026-92-g890cecc1

With the CI, it is now fairly easy to reproduce these errors. So I had a look and it turns out that the failure can be avoided by moving the `Seq(true, false).foreach { synth =>` outside of the test, like so:

```.diff
diff --git a/src/test/scala/neuroproc/unittests/TrueDualPortMemoryTester.scala b/src/test/scala/neuroproc/unittests/TrueDualPortMemoryTester.scala
index 682fd64..24e0fa5 100644
--- a/src/test/scala/neuroproc/unittests/TrueDualPortMemoryTester.scala
+++ b/src/test/scala/neuroproc/unittests/TrueDualPortMemoryTester.scala
@@ -81,9 +81,8 @@ class TrueDualPortMemoryTester extends AnyFlatSpec with ChiselScalatestTester {
     }
   }
 
-  it should "work in both Chisel and Verilog" in {
-    Seq(true, false).foreach { 
-      synth =>
+  Seq(true, false).foreach { synth =>
+    it should s"work in both Chisel and Verilog (synth=$synth)" in {
       test(TrueDualPortMemory(addrW, dataW, synth))
         .withAnnotations(Seq(VerilatorBackendAnnotation, WriteVcdAnnotation)) {
         dut => 
```

You were violating the implicit assumption that `test` will only be called once per test. I will have to talk to some people at Berkeley about that. We might decide to just add a warning or error if that happens or maybe add support for calling `test` more than once per test.

If you do not like the solution of moving the loop (it can make it harder to launch the test from an IDE since you are now _generating_ test cases), you can instead move the stimuli generation + output checks into their own function.

hansemandse, 2021-04-08:
> You were violating the implicit assumption that `test` will only be called once per test. I will have to talk to some people at Berkeley about that. We might decide to just add a warning or error if that happens or maybe add support for calling `test` more than once per test.

Ah, I wasn't aware of that. How come the tests ran just fine yesterday, though? Is it a new addition?
And how come the error it throws is the `VL_COVER...` stuff again?

ekiwi, 2021-04-08:
> Ah, I wasn't aware of that. How come the tests ran just fine yesterday, though? Is it a new addition?
> And how come the error it throws is the `VL_COVER...` stuff again?

I had the same assumption yesterday. It is just that the way the cover stuff is implemented now, it falls over the *.cpp files left over from the previous `test`.

We are considering officially support calling `test` multiple times. So if you wait a bit, we might have a patch for you next week.