ucb-bar/chiseltest #291: Feature request: Support peeks on ChiselEnum values

KasperHesse, 2021-04-07:
The new ChiselEnum in chisel3.experimental is only somewhat supported in chiseltest. While it is possible to poke and expect enum values, peeking enum values is not yet supported. Given that `expect` is supported, I would expect `peek` support to be relatively easy to implement. Doing so would make the enums a lot more useable.

ducky64, 2021-04-21:
This isn't actually super straightforward to support (needs additional APIs in base chisel), see the description for why in #111 and the overall enum issue #12.

I don't think it's something that's conceptually hard, but does require cross-cutting changes.

ekiwi, 2022-08-04:
fixed by https://github.com/ucb-bar/chiseltest/pull/542

closed 2022-08-04 by ekiwi