ucb-bar/chiseltest #301: Verilator backend fails intermittently

kammoh, 2021-05-14:
On latest published 0.5-SNAPSHOT and the default Verilator  coming with Fedora 34 (v4.106?)

verilator run _sometimes_ fails:
```
Elaborating design...
Done elaborating.
cd /home/kamyar/src/test/test_run_dir/Test_should_pass && verilator --cc Top.sv --assert -Wno-fatal -Wno-WIDTH -Wno-STMTDLY -O1 --top-module Top +define+TOP_TYPE=VTop +define+PRINTF_COND=!Top.reset +define+STOP_COND=!Top.reset -CFLAGS "-Wno-undefined-bool-conversion -O1 -DTOP_TYPE=VTop -DVL_USER_FINISH -include VTop.h" -Mdir /home/kamyar/src/test/test_run_dir/Test_should_pass --exe /home/kamyar/src/test/test_run_dir/Test_should_pass/Top-harness.cpp --coverage-user
- should return Top for Q=3329 *** FAILED ***
  java.lang.RuntimeException: Failed to find line `VL_COVER_INSERT(` in test_run_dir/Test_should_pass/VTop__Slow.cpp.
Please file an issue and include the output of `verilator --version`
  at chiseltest.legacy.backends.verilator.PatchCoverageCpp$.error(PatchCoverageCpp.scala:68)
  at chiseltest.legacy.backends.verilator.PatchCoverageCpp$.$anonfun$findLine$1(PatchCoverageCpp.scala:59)
  at scala.Option.getOrElse(Option.scala:201)
  at chiseltest.legacy.backends.verilator.PatchCoverageCpp$.findLine(PatchCoverageCpp.scala:59)
  at chiseltest.legacy.backends.verilator.PatchCoverageCpp$.replaceCoverage(PatchCoverageCpp.scala:32)
  at chiseltest.legacy.backends.verilator.PatchCoverageCpp$.$anonfun$apply$1(PatchCoverageCpp.scala:21)
  at chiseltest.legacy.backends.verilator.PatchCoverageCpp$.$anonfun$apply$1$adapted(PatchCoverageCpp.scala:20)
  at scala.collection.IterableOnceOps.foreach(IterableOnce.scala:563)
  at scala.collection.IterableOnceOps.foreach$(IterableOnce.scala:561)
  at scala.collection.AbstractIterable.foreach(Iterable.scala:919)
  ...
Execution took 3.77s
1 tests, 1 failed

===============================================
Total duration: 3.77s
1 failed

Failed:
- test.TopSpec:
  * Top should return Top for Q=3329 - java.lang.RuntimeException: Failed to find line `VL_COVER_INSERT(` in test_run_dir/Test_should_pass/VTop__Slow.cpp.
Please file an issue and include the output of `verilator --version`
===============================================
```
```
$ verilator --version
Verilator 4.108 2021-01-10 rev v4.106-158-g484b76e5b

$ uname -a
Linux fedora 5.11.18-300.fc34.x86_64 #1 SMP Mon May 3 15:10:32 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
```

removing test_run_dir content fixes the issue for a while, but running the test several times (without changing any of the design/test files) it shows up again. Very strange! I haven't yet seen this happening on macOS.

ekiwi, 2021-05-14:
Are you calling to `test` more than once in a single `scalatest` "unit"?
I.e., something like:

```.scala
it should "bla bla" in {
  test(new DUT) { ... }
  test(new DUT) { ... }
}
```

This can cause failures like the one you describe. I am hoping to fix that eventually, just have been pretty busy.

kammoh, 2021-05-14:
Well, kinda: the test structure calls `test` in a loop, but in the failing code, the loop was traversing only once:
```scala
  for (x <- Seq(123)) {
    it should s"pass" in {
      test(new Test(x)).withAnnotations(annos) { dut =>
        ....
      }
    }
  }
```
I'm also noticing that _sometimes_ the executed Verilator test does not match the last changes I made to the Chisel code! It rarely happens and I haven't figured out a way to reproduce it. I guess a lot of things (other than `chiseltest`) could be blamed for this including: the Scala build system (I'm using bloop/sbt), firrtl, or Verilator cpp generation and build.
Would it make sense to delete stale Verilator artifacts (including cpp and binary objects) before each run of `test`?
I also find it weird (and frankly messy!) that `chiseltest` does not use a subfolder for Verilator objects (`--Mdir`) and puts everything alongside the generated Verilog and firrtl. By default, Verilator puts the artifacts in a `obj_dir` subfolder.

ekiwi, 2021-05-14:
>  Would it make sense to delete stale Verilator artifacts (including cpp and binary objects) before each run of `test`?

I want to add that to `chiseltest` at some point, yes.

> I also find it weird (and frankly messy!) that `chiseltest` does not use a subfolder for Verilator objects (`--Mdir`) and puts everything alongside the generated Verilog and firrtl. By default, Verilator puts the artifacts in a `obj_dir` subfolder.

Feel free to make a PR to fix that. It's just legacy behavior.

> Well, kinda: the test structure calls `test` in a loop, but in the failing code, the loop was traversing only once:

Can you change the name to include `x`? So like: `it should s"pass_$x"`

kammoh, 2021-05-14:
> Can you change the name to include `x`? So like: `it should s"pass_$x"`
It actually included it when it at the time of failures and I removed it when trying to chase down the issue.

ekiwi, 2021-05-14:
> It actually included it when it at the time of failures and I removed it when trying to chase down the issue.

having unique test names is important so that you get unique folder.

I assume that cleaning up the Verilator stuff everytime before we call Verilator might fix your bugs. I am pretty busy until June, but I should have time now and then to review if you want to make a PR that makes Verilator create files in a subdirectory and also deletes any subdirectory left over from previous tests.

kammoh, 2021-05-14:
Thanks! I created https://github.com/ucb-bar/chisel-testers2/pull/303

ekiwi, 2021-05-20:
Please re-open this if the issue pops up again.

closed 2021-05-20 by ekiwi