ucb-bar/chiseltest #304: Extend peek/poke/expect to work with Bundle objects

richmorj, 2021-05-14:
When writing unit tests for Modules, which have, say, some standard bus interfaces, habitually I will create functions to which I pass the module and the parameters of some transaction to drive into a given interface on the Module. E.g.

```
  def busXfer(c: MyModuleWithBusInterface, data: Int) = {
    c.io.busIf.valid.poke(true.B)
    c.io.busIf.bits.poke(data.U)
    c.clock.step()
    //etc.
```
I can then create more complex tests, which perhaps send sequences of transfers to the bus interface of the design under test by using my busXfer() funtion, and passing the Module object to it.

Now, let's say I had more than one Module which had this bus interface and I wanted to reuse my useful functions on them all.  Right now, as far as I understand, I would not be able to do so as I would need different functions which took my new Module as a parameter e.g. 

```
    def busXfer(c: MyOtherModuleWithBusInterface, data Int) = {
    // ...
``` 

A feature which would allow a higher degree of code reuse would be if we could peek/poke/expect Bundles. We could then write functions to which we pass a Bundle object, which the function drives with peek/poke/expect.  The Bundle could then be connected to the design under test Module instance in some test harness.  The same test driver functions could then be used to drive any Module which had the given IO Bundle, making them portable and re-usable.

For anyone with experience with SystemVerilog I guess what I'm asking for is equivalent to driving a virtual interface, which is then connected to an actual interface on the DUT instance.  I'm only learning Chisel and chisel-testers and standby ready to be educated.

ducky64, 2021-05-14:
So there seems to be several questions here, which I may not be interpreting correctly (so correct me if I'm not!)

1. You're not limited to passing modules into functions (this may have been a limit of iotesters, but chiseltest was specifically designed to avoid that). Your `c` argument could be a Bundle type, though make sure you're passing in Bundles that are part of a DUT.
2. The idea of high-level 'transactions' called as functions is also supported. For example, see the [QueueTest](https://github.com/ucb-bar/chisel-testers2/blob/master/src/test/scala/chiseltest/tests/QueueTest.scala), which uses the high-level `.enqueueNow`, `.expectPeek`, and similar operations defined in [DecoupledDriver](https://github.com/ucb-bar/chisel-testers2/blob/master/src/main/scala/chiseltest/DecoupledDriver.scala). Note that this uses an implicit class pattern to extend a class defined externally, here to add test-specific methods to Decoupled defined in base Chisel. If you are defining the bundle under test, you can add test methods directly in it (though, whether that's stylistically great is a different question).
3. It is also possible to poke on Bundles directly with Bundle literals. See examples in [BundleLiteralsSpec](https://github.com/ucb-bar/chisel-testers2/blob/master/src/test/scala/chiseltest/tests/BundleLiteralsSpec.scala).
4. If you're asking to be able to define a hardware golden model in chiseltest that could be connected in a top-level testharness like any other module (mixed RTL / hardware construction and imperative programming semantics), that isn't supported. This seems like a hard problem (in general) that is difficult to do well.

richmorj, 2021-05-14:
Thanks for the super quick response. I will try passing a Bundle and connect that to my DUT Module somehow. To be honest, I assumed I could only peek/poke/expect on Module objects because I had not found an example of anything else.

richmorj, 2021-05-18:
Hi @ducky64 so I finally got back to this...  Basically, for maximum stimulus portability and code reuse, ideally I would like to use chisel-testers2 to peek/poke/expect _and clock.step()_ Bundle interfaces on my Modules, passing the hierarchical reference to the Module's io to my function(s).  So, modifying my example code above, I'd like to do this kind of thing:
```
class MyUnitTester extends FlatSpec with ChiselScalatestTester {
  behavior of "MyMod"

  def busXfer(b: MyBusInterfaceBundle, data: Int) = {
    b.valid.poke(true.B)
    b.bits.poke(data.U)
    b.clock.step() // <- uh oh
    //etc.
    }

  it should "drive the bus interface Bundle of MyMod" in {
    test(new MyMod) { dut =>
      busXfer(dut.io.b, 0x5a5a5a5a)
    }
  }
}

```
However, I can't _clock.step()_ a Bundle object.  It would be neat if you could associate a clock with a Bundle and step it though.

Anyway, I just discovered an example here in [ShiftRegisterTest](https://github.com/ucb-bar/chisel-testers2/blob/master/src/test/scala/chiseltest/tests/ShiftRegisterTest.scala) which I think gives me my solution.  Basically I just need to supply my Module's clock as a function argument as well as my its interface Bundle.  So, modifying my function from above again:

```
  def busXfer(b: MyBusInterfaceBundle, clk: Clock, data: Int) = {
    b.valid.poke(true.B)
    b.bits.poke(data.U)
    clk.step()
    //etc.
    }
```
Is this the answer? Please confirm and close the issue if so.

richmorj, 2021-05-20:
Unblocked! Thanks @ducky64

closed 2021-05-20 by richmorj

ducky64, 2021-05-26:
Sorry about the late reply - glad you got unblocked, but here's a bit more context:
The example you posted is one way to do this right now - pass in an explicit clock. It's not great, because you do need to pass in that extra argument, but it is simple.

However, you can also take a look at [DecoupledDriver's `setSourceClock`](https://github.com/ucb-bar/chisel-testers2/blob/a1afe0516a49f296cafb5aeab68b87413e96c2ef/src/test/scala/chiseltest/tests/ValidQueueTest.scala#L27) and `setSinkClock`. Under the hood, [this associates the wire with a clock in a global registry](https://github.com/ucb-bar/chisel-testers2/blob/master/src/main/scala/chiseltest/DecoupledDriver.scala#L18) (because DecoupledDriver is an implicit class and cannot maintain internal state). Longer term, this was meant to be able to automatically infer a clock from the circuit in the common / simple single-clock case, but that's waiting on the relevant FIRRTL analyses passes being upstreamed (@azidar ?).