ucb-bar/chiseltest #311: Problems with --coverage-user and chisel-testers2 master

oharboe, 2021-05-22:
Is there a test-case 

I've tried this with Verilator 4.028 and 4.202.

This command line creates a single .cpp file that has includes all the modules in the test.

```
/usr/bin/perl /home/oyvind/ascenium/verilator2/install/share/verilator/bin/verilator_includer ... VDoohickeyTestBench__ALL.cpp
```

```
// DESCRIPTION: Generated by verilator_includer via makefile
#define VL_INCLUDE_OPT include
#include "VDoohickeyTestBench.cpp"
#include "VDoohickey_Foo.cpp"
#include "VDoohickey_Bar.cpp"
```

In Verilator 4.028 this creates problems because static insertp() function is duplicated in each of the files above, causing error when __ALLcls.cpp is built.

In Verilator 4.202 there are different problems:

```
VDoohicekyTestBench__Slow.cpp: In function ‘void chisel_insertp(const char*, const char*, const char*, const char*, const char*, int, const char*, int, const char*, const string&, const char*, const char*, const char*, const char*, const char*, const char*)’:
VCacheInterfaceUnitTestBench__Slow.cpp:52:19: error: ‘_insertp’ is not a member of ‘VerilatedCov’
   52 |     VerilatedCov::_insertp(
[lots more errors deleted]
```

oharboe, 2021-05-22:
There are various problems with Ubuntu 20.04, so I had to create the pull request below to get chisel-testers2 going on Ubuntu 20.04 and with Verilator 4.202:

To reproduce:

1. Checkout https://github.com/ucb-bar/chisel-testers2/pull/313
2. Build Verilator 4.202
3. Run the command below on Ubuntu 20.04:

```
sbt test
```

Snippet of result:

```
[deleted]
In file included from VVerilatorBasicTests_Anon__ALL.cpp:4:
VVerilatorBasicTests_Anon__Slow.cpp: In function ‘void chisel_insertp(const char*, const char*, const char*, const char*, const char*, int, const char*, int, const char*, const string&, const char*, const char*, const char*, const char*, const char*, const char*)’:eltest.tests.ThreadSafetyTest 10s
VVerilatorBasicTests_Anon__Slow.cpp:49:19: error: ‘_insertp’ is not a member of ‘VerilatedCov’
   49 |     VerilatedCov::_insertp(
      |                   ^~~~~~~~
In file included from ./VVerilatorBasicTests_Anon.h:12,
                 from <command-line>:
VVerilatorBasicTests_Anon__Slow.cpp: In member function ‘void VVerilatorBasicTests_Anon::__vlCoverInsert(uint32_t*, bool, const char*, int, int, const char*, const char*, const char*, const char*)’:
VVerilatorBasicTests_Anon__Slow.cpp:36:31: error: ‘_inserti’ is not a member of ‘VerilatedCov’
   36 |     VL_IF_COVER(VerilatedCov::_inserti(countp); VerilatedCov::_insertf(__FILE__, __LINE__); \
      |                               ^~~~~~~~
install/share/verilator/include/verilated_cov.h:44:10: note: in definition of macro ‘VL_IF_COVER’
[deleted]
[error] Failed: Total 180, Failed 43, Errors 0, Passed 137, Ignored 4, Canceled 8
[error] Failed tests:
[error] 	chiseltest.backends.verilator.VerilogBlackBoxTest
[error] 	chiseltest.backends.verilator.VerilatorCoverageTest
[error] 	chiseltest.backends.verilator.VerilatorBasicTests
[error] 	chiseltest.backends.verilator.VerilatorCoverageTests
[error] 	chiseltest.backends.verilator.VerilatorClockPokeTest
[error] 	chiseltest.backends.verilator.VerilatorCachingTests
[error] 	chiseltest.experimental.tests.VerilatorCachingTests
[error] 	chiseltest.experimental.tests.NegativeInputValuesTest
[error] (Test / test) sbt.TestsFailedException: Tests unsuccessful
[error] Total time: 32 s, completed May 22, 2021, 1:12:26 PM
```

ekiwi, 2021-08-13:
We merged the Verilator 4.202 support. Please open a new issue if you have any further issues.

closed 2021-08-13 by ekiwi