ucb-bar/chiseltest #316: Build failure with Verilator with two modules(probably) in some cases

oharboe, 2021-05-26:
This problem is not reproducible with a single module test.

#315  is the smallest failure case I could produce. It uses a BlackBox for Verilog, but I've seen it with Chisel only code too. #315 just happened to be the smallest standalone testcase that I can easily publish.

```
2021-05-26T06:05:01.8437664Z In file included from VDoohickey__ALLcls.cpp:4:
2021-05-26T06:05:01.8440981Z VDoohickey_Doohickey.cpp:34:13: error: redefinition of ‘void chisel_insertp(const char*, const char*, const char*, const char*, const char*, int, const char*, int, const char*, const string&, const char*, const char*, const char*, const char*, const char*, const char*)’
2021-05-26T06:05:01.8442818Z    34 | static void chisel_insertp(
2021-05-26T06:05:01.8443436Z       |             ^~~~~~~~~~~~~~
2021-05-26T06:05:01.8509088Z In file included from VDoohickey__ALLcls.cpp:3:
2021-05-26T06:05:01.8510926Z VDoohickey.cpp:38:13: note: ‘void chisel_insertp(const char*, const char*, const char*, const char*, const char*, int, const char*, int, const char*, const string&, const char*, const char*, const char*, const char*, const char*, const char*)’ previously defined here
2021-05-26T06:05:01.8512650Z    38 | static void chisel_insertp(
2021-05-26T06:05:01.8513269Z       |             ^~~~~~~~~~~~~~
2021-05-26T06:05:01.8525668Z In file included from VDoohickey__ALLcls.cpp:5:
2021-05-26T06:05:01.8528064Z VDoohickey_DualPortNewDataRAM__W40_A4.cpp:33:13: error: redefinition of ‘void chisel_insertp(const char*, const char*, const char*, const char*, const char*, int, const char*, int, const char*, const string&, const char*, const char*, const char*, const char*, const char*, const char*)’
2021-05-26T06:05:01.8529680Z    33 | static void chisel_insertp(
2021-05-26T06:05:01.8530441Z       |             ^~~~~~~~~~~~~~
2021-05-26T06:05:01.8566543Z In file included from VDoohickey__ALLcls.cpp:3:
2021-05-26T06:05:01.8568484Z VDoohickey.cpp:38:13: note: ‘void chisel_insertp(const char*, const char*, const char*, const char*, const char*, int, const char*, int, const char*, const string&, const char*, const char*, const char*, const char*, const char*, const char*)’ previously defined here
2021-05-26T06:05:01.8570078Z    38 | static void chisel_insertp(
2021-05-26T06:05:01.8570662Z       |             ^~~~~~~~~~~~~~
2021-05-26T06:05:02.0966932Z cc1plus: warning: unrecognized command line option ‘-Wno-undefined-bool-conversion’
2021-05-26T06:05:02.1051362Z make: *** [/usr/share/verilator/include/verilated.mk:182: VDoohickey__ALLcls.o] Error 1
2021-05-26T06:05:02.1052267Z make: *** Waiting for unfinished jobs....
2021-05-26T06:05:02.2830680Z Writing Coverage..
```

ekiwi, 2021-05-26:
Thanks for the explanation @oharboe. This sound like a bug in code that I wrote. Pretty busy right now, but I hope I will have some time in June to look into this.

ekiwi, 2021-08-13:
I hope this will be fixed by: https://github.com/ucb-bar/chisel-testers2/pull/363

Thanks for the example you provided and please let me know if there are still failing tests for you.

closed 2021-08-13 by ekiwi