ucb-bar/chiseltest #318: BlackBox Verilator Error

Talha-Ahmed-1, 2021-05-27:
Testing BlackBox by using Verilator backend and generating its vcd but it gives below error.

`java.lang.AssertionError: assertion failed: verilator command failed on circuit Practice in work dir test_run_dir/Practice`
`[info] PracticeTests: [info] - Practice *** FAILED *** [info] java.lang.AssertionError: assertion failed: verilator command failed on circuit Practice in work dir test_run_dir/Practice [info] at scala.Predef$.assert(Predef.scala:223) [info] at chisel3.assert$.apply(Assert.scala:70) [info] at chiseltest.legacy.backends.verilator.VerilatorExecutive$.start(VerilatorExecutive.scala:111) [info] at chiseltest.defaults.package$.createDefaultTester(defaults.scala:24) [info] at chiseltest.ChiselScalatestTester$TestBuilder.apply(ChiselScalatestTester.scala:33) [info] at SPI.PracticeTests.$anonfun$new$1(PracticeTests.scala:17) [info] at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:23) [info] at org.scalatest.OutcomeOf.outcomeOf(OutcomeOf.scala:85) [info] at org.scalatest.OutcomeOf.outcomeOf$(OutcomeOf.scala:83) [info] at org.scalatest.OutcomeOf$.outcomeOf(OutcomeOf.scala:104)`

Talha-Ahmed-1, 2021-05-27:
[Link to code](https://github.com/Talha-Ahmed-1/SPI-Chisel/blob/main/src/main/scala/SPI/Practice.scala)

kammoh, 2021-05-27:
`Procedural assignment to wire`.. just need to fix the inlined Blackbox Verilog!
I created a  PR for you.

closed 2021-05-27 by ekiwi

kammoh, 2021-05-28:
Could you double check verilator is installed on the machine and is in the path? What’s the output of `verilator —version`?

Talha-Ahmed-1, 2021-05-28:
Yes verilator is installed I test many other Modules using chiseltest and verilator it only shows error in BlackBox testing.
But now after pull request it doesn't even test simple modules.
`verilator --version = Verilator 3.904 2017-05-30 rev verilator_3_904`

kammoh, 2021-05-28:
Verilator 3.904 is not a supported version. Please see the the README for a list of currently supported versions. `v4.202` is also supported in the latest `0.5-SNAPSHOT` release.

kammoh, 2021-05-28:
I think the reported error should be improved though.

ekiwi, 2021-05-28:
> I think the reported error should be improved though.

We could check for a supported version and refuse to run if we cannot find one. However, I have my suspicion that we are going to get people who will tell us that they are running an unsupported version and "it works for me"..