ucb-bar/chiseltest #345: FST traces in verilator

nrother, 2021-07-26:
Verilator allows to write [FST trace files](https://veripool.org/guide/latest/faq.html#how-do-i-generate-fst-waveforms-traces-in-c-or-systemc) using the `--trace-fst` option. It would be nice if this would be exported through chiseltest.

I did run any experiments on it, but FST files are supposed to be much smaller than VCD equivalents (which tent to be shockingly large).

ekiwi, 2021-07-26:
FST traces will be available once we get https://github.com/ucb-bar/chisel-testers2/pull/334 and https://github.com/ucb-bar/chisel-testers2/pull/337 merged.

nrother, 2021-07-27:
Ah, that sounds good!
(I guess I'll leave this open as a reminder, but feel free to close, if you like)