ucb-bar/chiseltest #347: Output directory is not created if it is a symlink

nrother, 2021-07-27:
On my machine, my home-folder is on a network share. To speed things up when running tests, I created a symlink from `test_run_dir` to `/tmp/testrun`, which is on a RAM disk. After a reboot, the directory in `/tmp` is gone, creating a dangling symlink.

This situation causes chiseltest to fail. It would be nice if this situation would be detected and the *destination* of the symlink would be created automatically (as it is the case if the directory is just missing).

(To be honest, this is a pretty exotic case, and I'm not so sure if is is worth special-casing this in chiseltest...)

ekiwi, 2021-07-27:
I don't think chiseltest should fix dangling symlinks. If you think the current error message is confusing, we would be open to improving that though.

nrother, 2021-07-29:
I agree that this is probably out of scope for chiseltest; there also seems to be no simple way for a "mkdir though symlinks".

The current error message (using Verilator as a backend) is somewhat confusing:

```
Elaborating design...
Done elaborating.
createFile error: java.nio.file.NoSuchFileException: test_run_dir/ToplevelClusterController_should_pass_the_integration_test/sim_api.h
[info] ToplevelClusterControllerTest:
[info] ToplevelClusterController
[info] - should pass the integration test *** FAILED ***
[info]   java.nio.file.NoSuchFileException: test_run_dir/ToplevelClusterController_should_pass_the_integration_test/sim_api.h
[info]   at java.base/sun.nio.fs.UnixException.translateToIOException(UnixException.java:92)
[info]   at java.base/sun.nio.fs.UnixException.rethrowAsIOException(UnixException.java:111)
[info]   at java.base/sun.nio.fs.UnixException.rethrowAsIOException(UnixException.java:116)
[info]   at java.base/sun.nio.fs.UnixFileSystemProvider.newByteChannel(UnixFileSystemProvider.java:219)
[info]   at java.base/java.nio.file.spi.FileSystemProvider.newOutputStream(FileSystemProvider.java:478)
[info]   at java.base/java.nio.file.Files.newOutputStream(Files.java:224)
[info]   at java.base/java.nio.file.Files.copy(Files.java:3144)
[info]   at chiseltest.legacy.backends.verilator.CopyVerilatorHeaderFiles$.apply(CopyVerilatorHeaderFiles.scala:29)
[info]   at chiseltest.legacy.backends.verilator.VerilatorExecutive$.start(VerilatorExecutive.scala:52)
[info]   at chiseltest.defaults.package$.createDefaultTester(defaults.scala:24)
[info]   ...
```

ekiwi, 2021-08-13:
Sorry, but I think this is a won't fix. Feel free to open a PR if you have an idea on how to make the error message better to understand.

closed 2021-08-13 by ekiwi